homieiot/homie-esp8266 #175: Upgrading PlatformIO to v2

lemining, 2016-10-15:
Hi!

Amazing work on the library and the whole platform. I have spent some time today to see if I can get use out of my esp8266.

I couldn't get examples to work and then I realized that I was running 1.5 (as this is the latest on PlatformIO).

Could you please update?
http://platformio.org/lib/show/555/Homie

Kind regards,
Luke

lemining, 2016-10-15:
Closing! I just realized that 2 is in development! 

Sorry!

closed 2016-10-15 by lemining

claybar, 2016-10-15:
Rather than referencing the library via its number (555) you can use the
git version directly in platformio. Just put the git link in the ini file:
https://github.com/marvinroger/homie-esp8266.git

lemining, 2016-10-15:
Indeed. For reference this is my platformio.ini file to get v2 going.

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/en/stable/projectconf.html

[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_deps =
  git+https://github.com/marvinroger/homie-esp8266.git#develop
  git+https://github.com/bblanchon/ArduinoJson.git#v5.6.7
  git+https://github.com/marvinroger/async-mqtt-client.git#master
  git+https://github.com/thomasfredericks/Bounce2.git#master
  git+https://github.com/me-no-dev/ESPAsyncTCP.git#master
`