homieiot/homie-esp8266 #176: Homie Configurator didn't go to next step

Tokandol, 2016-10-15:
Homie configurator did not go to next step, when I checked "Use MQTT authentication".
But, if not checked, can go to next step. I tested Web UI and app, and result is same.
I couldn't find any issues about.  Now I'm testing with release 2.0.0 and using CloudMQTT broker.

[image]

So I trying to debug using Chrome on PC, there are some errors below when pushed next button.
[image]
[image]

marvinroger, 2016-10-15:
It should be fixed now. Can you try again?

Tokandol, 2016-10-16:
Thanks Marvin.

It works good. But there are problem " Invalid or too big JSON" in serial monitor.  It seems same second issue in [#178.](https://github.com/marvinroger/homie-esp8266/issues/178). 
 Below is capture of structure of data.

[image]

marvinroger, 2016-10-16:
Can you try with the latest git?

Tokandol, 2016-10-17:
Hi Marvin

It work good without any error. But my device did not work normally. 
It repeat endlessly to connect MQTT and disconnect MQTT.  

[image]

Tokandol, 2016-10-17:
The reason of repetition is I have a another device using same Device Id.  So MQTT is disconnected and Homie trying to reconnect.

closed 2016-10-17 by Tokandol

marvinroger, 2016-10-17:
Great!

alco28, 2016-10-17:
double confirmed that it works :) thanks for your effort on all! I'm just starting to learn the capabilities of an ESP8266 with your stuff!