homieiot/homie-esp8266 #178: 2 Bugs? Config and $online after todays commit

Gulaschcowboy, 2016-10-15:
Hi Marvin,

first off all: Great project, I love it.

Two things I discovered after I did a git pull today.

1. to devices/<id>/$online alsways false is submitted
so I thought, ok, let's do a reset. But I discovered another thing:

2. Config always reports: "There was an error while sending the configuration: Invalid or too big JSON
Please retry or report if it looks like a bug. "
I used latest app version and also the online app of you.

Thanks,
Alex

Gulaschcowboy, 2016-10-15:
Update:
I tried curl method and found this behaviour:

Sending this config it works:
{
  "name": "bee2",
    "wifi": {
    "ssid": "my-ssid",
    "password": "my-pass"
     },
     "mqtt": {
     "host": "omd.atze.lan",
     "port": 1883
     },
     "ota": {
     "enabled": true
      }
}

If I add auth credentials like:
{
  "name": "bee2",
    "wifi": {
    "ssid": "my-ssid",
    "password": "my-pass"
     },
     "mqtt": {
     "host": "omd.atze.lan",
     "port": 1883
     "auth": true,
     "username": "my-mqttuser",
     "password": "my-mqttpass"
     },
     "ota": {
     "enabled": true
      }
}

I get the response: {"success":false,"error":"Invalid or too big JSON"}

marvinroger, 2016-10-15:
I know what's wrong with the second issue, it's because ArduinoJson tries to change the content of the internal buffer of an Arduino String, which is wrong. That's not actually an issue from ArduinoJson, though.

What do you mean by `two devices//$online always false are submitted`? Like, at startup? Is an `$online true` submitted or not at all?

Gulaschcowboy, 2016-10-15:
Hi Marvin,

after recompiling and flashing newest github version the device always posts "false" to devices/<dev-id>/$online

What is solution or workaround to issue 1?

marvinroger, 2016-10-15:
I did not fix anything yet, I'll tell you when it's done.

The device never post `false` to `$online`, but the broker implicitely sends this when the device is disconnected from the TCP socker. Or, you use `prepareToSleep()`.

Can you post your sketch?

Gulaschcowboy, 2016-10-16:
[homie-Bienen1.txt](https://github.com/marvinroger/homie-esp8266/files/531508/homie-Bienen1.txt)

Sorry, couldn't post inline, as github api eats some code parts

marvinroger, 2016-10-16:
Well, your second JSON is actually invalid (you forgot a `,`) after the `1883`. I tried this JSON with the `,` and it works fine.

marvinroger, 2016-10-16:
Can you try with the latest git? The second issue should be resolved. I am not able to reproduce the first issue, though, so try to send an empty retained message to the topic on which you receive the two `false`, so it will clean the old retained message.

Gulaschcowboy, 2016-10-16:
Hi Marvin,
yes, first issue is solved. And I noticed now, that the default prefix changed from devices/ to homie/. That's why I always saw false in devices/<node>/$online and why the homie-OTA App didn't recognize it anymore.
Sorry, I missed that change. Thank you!

closed 2016-10-16 by Gulaschcowboy