homieiot/homie-esp8266 #181: Issue with setMaxTopicLength with dev branch

AlexValois, 2016-10-16:
Hi,

First of all, thank you VERY MUCH for your work. I really like Homie library ! 

I have just replaced the master (still v1) with the v2 release. After adapting code as mentioned in documentation, I still have an issue with AsyncMQTTClient library when I try to compile the examples or my own sketchs.

The error message is:
```
Arduino: 1.6.8 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

/root/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootNormal.cpp: In member function 'virtual void HomieInternals::BootNormal::setup()':
/root/Arduino/libraries/homie-esp8266-develop/src/Homie/Boot/BootNormal.cpp:497:44: error: 'class AsyncMqttClient' has no member named 'setMaxTopicLength'
   _interface->mqttClient->setKeepAlive(10).setMaxTopicLength(MAX_MQTT_TOPIC_LENGTH);
                                            ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

``` 

I have commented following line in BootNormal.cpp (line 497) and it seems to work fine:
`_interface->mqttClient->setKeepAlive(10).setMaxTopicLength(MAX_MQTT_TOPIC_LENGTH);

I have looked for a new version (or a dev branch for async-mqtt-client) of AsycMQTTClient lib but I did not find any. May I have miss something or is it a bug ?

Thank you

Alex

marvinroger, 2016-10-16:
This was implemented 6 days ago, in https://github.com/marvinroger/async-mqtt-client/commit/3f8cc2fe63dd9911e57a83ca0f2a8ffc8805bc16

Download the latest [.zip tree](https://github.com/marvinroger/async-mqtt-client/archive/master.zip) and you shoud be fine!

marvinroger, 2016-10-16:
And glad you like the project. 😉

AlexValois, 2016-10-16:
Ok thanks. I will make a try.

marvinroger, 2016-10-25:
@AlexValois any update?

AlexValois, 2016-10-25:
With latest versions of async and homie-esp8266, it works fine.

Thank you

closed 2016-10-25 by AlexValois