homieiot/homie-esp8266 #183: OTA of ESP8266-07 (1M flash) after commit 6 days ago

Gulaschcowboy, 2016-10-16:
Hi Marvin,

the OTA of my ESPs 07 (1M flash, 512K SPIFFS) doesn't work anymore after your commit 6 days ago. Before it was possible. Now it fails with serial output like:

Receiving OTA payload (349680/349680)...
�� OTA failed

So the ~350K actually should fit into the SPIFFS
What do you think, are the program/flash sizes getting to small for OTA Updates?
My NodeMCUs with 4M/3M SPIFFS can be updated...

Thank you,
Alex

marvinroger, 2016-10-17:
Look at the files this PR change: https://github.com/marvinroger/homie-esp8266/pull/174

Just add this:
```c++
String error();
Update.printError(error);
_interface->logger->logln(error);
```
after `_interface->logger->logln(F("✖ OTA failed"));`

And you'll see the actual error, which will tell you what the problem is. I indeed think the flash size is getting too small.

Gulaschcowboy, 2016-10-17:
Thanks, I tried:

--- BootNormal.cpp.bak  2016-10-17 11:33:50.032120664 +0200
+++ BootNormal.cpp      2016-10-17 11:34:35.648674004 +0200
@@ -249,7 +249,10 @@
           _flaggedForReboot = true;
         } else {
[compile_error.txt](https://github.com/marvinroger/homie-esp8266/files/532917/compile_error.txt)

           _interface->logger->logln(F("✖ OTA failed"));
-          _interface->logger->logln(F("Triggering OTA_FAILED event..."));
+         String error();
+         Update.printError(error);
+         _interface->logger->logln(error);
+         _interface->logger->logln(F("Triggering OTA_FAILED event..."));
           _interface->event.type = HomieEventType::OTA_FAILED;
           _interface->eventHandler(_interface->event);
         }

But it throws a compile error, see txt.
As I am not a coder, I cant really resolve it myself. Sorry.

Flash limitation of the ESP 07 versus Homie's OTA would really be a bad showstopper for me using homie. I chose the 07 because of it's external antenna, as I'm monitoring my remote beehives with it. As it is remote and under a bee hive, you don't wanna sit there with your laptop flashing it. Bees don't like serial flashing.

Gulaschcowboy, 2016-10-17:
OK I just saw now, that you committed changes again. After git pull some errors less: 
[compile_error.txt](https://github.com/marvinroger/homie-esp8266/files/533016/compile_error.txt)

marvinroger, 2016-10-17:
My bad, add at the top of BootNormal.hpp: `#include <StreamString.h>`

Ans replace the code I told you to add by this one:

```c++
StreamString error;
Update.printError(error);
_interface->logger->println(error);
```

marvinroger, 2016-10-17:
If your issue is indeed space, the solution is pretty simple: lower your SPIFFS size.

The OTA does not flash the sketch to the SPIFFS. The SPIFFS is not touched during an OTA update. You said `1M flash, 512K SPIFFS`, so there's 512K for SPIFFS and 1024-512 = 512K for program space. If the sketch is 350K, there's only 512 - 350 = 162K available for an OTA, which is not enough. If you don't use SPIFFS in your program, select the `1M (64K SPIFFS)` preset. Therefore, you have 1024 - 64 = 960K for program space (which is enough for the UI bundle and the config file). If your current program is 350K, then there are 960-350 = 610K available for an OTA. Which is enough!

This is a simplified schema, of course, but you get the idea.

Gulaschcowboy, 2016-10-17:
Hi Marvin, yes, that was the case:
✖ OTA failed
ERROR[4]: Not Enough Space
OK, I got it. I thought the OTA is saved within the SPIFFS, but I was wrong. Will give it a shot!

Thanks man!

Gulaschcowboy, 2016-10-17:
Thank you, it works.

closed 2016-10-17 by Gulaschcowboy