homieiot/homie-esp8266 #193: Properties settable by default

claybar, 2016-10-23:
I have set up a node with two properties.  Pertinent code looks like:
```
HomieNode onboardNode("atmospheric", "temperature,humidity");

void setup()
{
  onboardNode.advertise("temperature");
  onboardNode.advertise("humidity");
  
  Homie.setup();
}
```

When connected to MQTT this seems to have set both properties as settable:
```
devices/blah/$homie 2.0.0
devices/blah/$implementation esp8266
devices/blah/atmospheric/$type temperature,humidity
devices/blah/atmospheric/$properties temperature:settable,humidity:settable
```

Is this expected?  I thought the default would be non-settable, with settable being set if `.settable(handler)` was added to the `.advertise("")` call.

marvinroger, 2016-10-23:
It's not intended, so it sounds like a bug!

marvinroger, 2016-10-24:
I don't see this behavior, it works correctly on my side. Can you pull the latest git and try again?

claybar, 2016-10-25:
Commit 7f7d373 seems to have fixed the problem.  My two property example is now correctly setting both as non-settable by default.

A quick experiment with one set as settable with a handler also reports correctly via mqtt.

Closing.

closed 2016-10-25 by claybar

marvinroger, 2016-10-25:
By default booleans are false, so that's weird. Anyway, glad it fixed the problem.

graugans, 2016-10-25:
@marvinroger where in the standard is it stated boolean is are false as default?

marvinroger, 2016-10-25:
It looks like my C# classes took over my limited C/C++ knowledge. I happily admit i'm wrong! Thanks for the heads up, @graugans