homieiot/homie-esp8266 #194: MQTT OTA not working?

prijswij, 2016-10-23:
I'm using homie 2.0.0.
I have a RPi with mosquitto installed.
On the pi I also installed homie-ota.

When I push an firmware update, I see the firmware binary on the MQTT queue "homie/devicename/$implementation/ota/firmware"
Then on "homie/devicename/$ota" is see the new firmware version.

But nothing happends? I'm monitoring "homie/devicename/$implementation/ota/status" but no message appears.

What am I missing?

adityat90, 2016-10-23:
I too am facing the same issue.
Using https://github.com/jpmens/homie-ota
And this project at commit e5205887c3e231dece8ddc9c72a2460961198b31

My code has the following
```
/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */
```
When the firmware.bin file is uploaded to mqtt queue, the esp8266 prints the following in the Serial log
```
✴ OTA available (version 0.0.2)  
Subscribing to OTA payload...
```

But nothing after that.
Is there anything that I'm missing?

mrpace2, 2016-10-23:
@prijswij Are you using the latest `homie-esp8266` from the `develop` branch and the latest greatest `homie-ota`? If so, can you please provide an MQTT log showing your device's boot sequence and also the messages `homie-ota` publishes during update?

    mosquitto_sub -v -t "homie/devicename/#"

There should be `homie/devicename/$implementation/ota/status` messages. Also, if possible, take a look at the serial output, or post it here.

mrpace2, 2016-10-23:
@adityat90 This is probably unrelated. Your `homie-esp8622` is old. It uses a different topic for firmware updates than the latest `homie-ota`. Can you upgrade your `homie-esp8622` and your `homie-ota` to the latest commits from gihub?

adityat90, 2016-10-23:
@mrpace2 Seems like that did the trick! Thanks a lot!!!
I got it to work. Not too sure how though

Made a couple of changes on the way
- Used this commit: 7bd50d7699896de171a16b9d3adb4e1b9f71dfcb
- Changed the default base topic to homie/
- Renamed my device from xxx-xxx-xx to a shorter name xx-xx

mrpace2, 2016-10-23:
@adityat90 Thanks. Glad it works now. BTW, a couple of hours ago I commited a PR to `homie-ota` fixing the order of MQTT messages for Homie 2. The PR has been merged. I suggest that you update `homie-ota` once again if you pulled before the merge. Otherwise you'd need to press `Go!` twice.

adityat90, 2016-10-24:
@mrpace2 Yes i noticed that too. Wasn't too sure of why it was happening, but since it was working on every second press of `Go!` I just let it be for then and called it a night.  
I'm loving this project.
Thanks again for all your help!

marvinroger, 2016-10-24:
@adityat90 you can remove the magic markers, it is now done behind the scene when you call `Homie_setFirmware()`

adityat90, 2016-10-24:
@marvinroger fabulous! I'll try this soon. A bit busy getting node red integrated right now, but will come back again. If there's anything more that needs testing, let me know. I'd be glad to assist with anything you'd need.

prijswij, 2016-10-24:
I got it working, I think it might be a difference in version (build). I have homie 2.0.0 running and did do an update yesterday of homie-ota. I think after pushing the bin to homie/devicename/$implementation/ota/payload instead of firmware it worked, could that be possible? 

I'll do a git pull on homie and homie-ota tonight and do some more testing.

prijswij, 2016-10-24:
Update: Using the latest of the latest of both homie and homie-ota, the sonoff is updating OTA.
- what i do not see is a status update on the queue
- afterwards, the sonoff keeps rebooting after startup. after resetting the sonoff and configuring it with the "old" values (same device name etc.) it keeps rebooting after startup.
When I reset, configure with a new device id, it works fine?

prijswij, 2016-10-24:
Not sure why... but now I see 'status' updates on the queue....

The only thing I did differently is I initiated everything using mosquitto_pub.
 - mosquitto_pub -h localhost -t "homie/sonofftest/\$ota" -m "1.2.1"
 - mosquitto_pub -h localhost -t "homie/sonofftest/\$implementation/ota/firmware" -f /home/hota/homie-ota/firmwares/itead-sonoff-1.2.1.bin

=== LOG START ===
homie/sonofftest/$ota 1.2.1
homie/sonofftest/$implementation/ota/status 202
homie/sonofftest/$implementation/ota/firmware ............
homie/sonofftest/$implementation/ota/status 206 1409/346288
......................
homie/sonofftest/$implementation/ota/status 206 346288/346288
homie/sonofftest/$implementation/ota/status 200
homie/sonofftest/$homie 2.0.0
homie/sonofftest/$implementation esp8266
homie/sonofftest/relay/$type switch
homie/sonofftest/relay/$properties on:settable
homie/sonofftest/$name test sonoff
homie/sonofftest/$localip 192.168.1.32
homie/sonofftest/$uptime/interval 120
homie/sonofftest/$fw/name itead-sonoff
homie/sonofftest/$fw/version 1.2.1
homie/sonofftest/$implementation/config {"name":"test sonoff",............,"device_id":"sonofftest"}
homie/sonofftest/$implementation/version 2.0.0
homie/sonofftest/$implementation/ota/enabled true
homie/sonofftest/$online true
homie/sonofftest/$signal 100
homie/sonofftest/$uptime/value 7
homie/sonofftest/relay/on true
homie/sonofftest/$implementation/ota/status 304

marvinroger, 2016-10-25:
@prijswij don't forget that the v2 is still in development, so topics are always subject to changes. So yes, it was probably messed up because something changed between two updates.

closed 2016-11-14 by marvinroger