homieiot/homie-esp8266 #198: Unable to configure using HTTP UI

kylegordon, 2016-10-26:
Hi,

When using v1.5.0 via PlatformIO, the UI fails to proceed beyond the MQTT tab. 

The serial output only outputs the following...

✔ Wi-Fi scan completed
Received heart request
Received device info request
Received networks request
Triggering Wi-Fi scan...
✔ Wi-Fi scan completed
Triggering Wi-Fi scan...

... and so on.

The sketch is just the example TemperatureSensor.ino in a fresh PlatformIO project, with the following platformio.ini

```
[env:nodemcu]
platform = espressif8266
framework = arduino
board = nodemcu
lib_deps = Homie
```

PlatformIO confirms that v1.5.0 of Homie is being downloaded and used.

When wifi is connected, the node shows up in both the web UI, downloaded web UI, and the PhoneGap app as expected, but in all cases I can get as far as the MQTT tab, enter the IP of the broker, but no further. Hitting Next does nothing.

Any thoughts? Thanks!

marvinroger, 2016-10-26:
Which UI do you use? Can you link it?

kylegordon, 2016-10-26:
Hi, I tried 3 UIs... http://setup.homie-esp8266.marvinroger.fr/ , https://codeload.github.com/marvinroger/homie-esp8266-setup/zip/gh-pages and also the Android apk from https://build.phonegap.com/apps/1906578/share

marvinroger, 2016-10-26:
Use this one for the v1.5.0: http://marvinroger.github.io/homie-esp8266/ and the docs from the [master branch](https://github.com/marvinroger/homie-esp8266/tree/master). I know, it's messy, sorry!

marvinroger, 2016-10-26:
The UI you used is for the upcoming v2 😉

kylegordon, 2016-10-26:
Woo! Connected, thank you. No problem at all. This might help others as well :-D 

Closes #198

closed 2016-10-26 by kylegordon