homieiot/homie-esp8266 #201: Handler

manelmg, 2016-10-30:
Hi Marvin,
I trying to use some handler event, but I am not able to execute the handler function.

I am using the LightOnOff.ino, I can see all information is published on my MQTT broker, but when I publish a new message in a settabled property the lightOnHandler is not executed.

When the EPS8266 is running in normal mode I see all messages like:

homie/train01/$homie --> 2.0.0
homie/train01/$implementation --> esp8266
homie/train01/light/$type --> switch
homie/train01/light/$properties --> on:settable
homie/train01/$name --> trainTest
 ...

Then I try to publish a new message, something like this:

mosquitto_pub -d -t homie/train01/light/on -m true
Client mosqpub/2086-rpi sending CONNECT
Client mosqpub/2086-rpi received CONNACK
Client mosqpub/2086-rpi sending PUBLISH (d0, q0, r0, m1, 'homie/train01/light/on', ... (4 bytes))
Client mosqpub/2086-rpi sending DISCONNECT

mosquitto_pub -d -t homie/train01/light/on -m false
Client mosqpub/2086-rpi sending CONNECT
Client mosqpub/2086-rpi received CONNACK
Client mosqpub/2086-rpi sending PUBLISH (d0, q0, r0, m1, 'homie/train01/light/on', ... (5 bytes))
Client mosqpub/2086-rpi sending DISCONNECT

The property is changed, but the lightOnHandler function is not executed (I don't see any message on serial monitor

Thanks in advance,
M

marvinroger, 2016-10-30:
Hi, just add a `/set` in your topic, and it will work. Don't forget to carefully read the Homie convention. 😊

manelmg, 2016-10-30:
Upps, apologize for this, I didn't read all Homie convention :(
Thanks for you quick response.

closed 2016-10-31 by manelmg