homieiot/homie-esp8266 #202: Strange compilation errors

LarsAC, 2016-10-30:
Hello,

I am using Arduino IDE 1.6.12 and homie-esp8266 v1.5.0. Library registration shows successfully under installed libraries.

When I try to compile a sample sketch from the homepage, I see strange errors which seem to suggest that the homie library was not correctly setup:

```
Arduino: 1.6.12 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

mqtt_relay:7: error: 'HomieRange' does not name a type
 bool lightOnHandler(const HomieRange& range, const String& value) {
                           ^
mqtt_relay:7: error: ISO C++ forbids declaration of 'range' with no type [-fpermissive]
 bool lightOnHandler(const HomieRange& range, const String& value) {
                                       ^
mqtt_relay:7: error: 'HomieRange' does not name a type
 bool lightOnHandler(const HomieRange& range, const String& value) {
                           ^
mqtt_relay:7: error: ISO C++ forbids declaration of 'range' with no type [-fpermissive]
 bool lightOnHandler(const HomieRange& range, const String& value) {
                                       ^
/Users/larsvonwedel/Documents/Arduino/mqtt_relay/mqtt_relay.ino: In function 'bool lightOnHandler(const int&, const String&)':
mqtt_relay:12: error: 'class HomieNode' has no member named 'setProperty'
   lightNode.setProperty("on").send(value);
             ^
mqtt_relay:13: error: no match for 'operator<<' (operand types are 'HardwareSerial' and 'const char [10]')
   Serial << "Light is " << (on ? "on" : "off") << endl;
          ^
mqtt_relay:13: error: 'endl' was not declared in this scope
   Serial << "Light is " << (on ? "on" : "off") << endl;
                                                   ^
/Users/larsvonwedel/Documents/Arduino/mqtt_relay/mqtt_relay.ino: In function 'void setup()':
mqtt_relay:20: error: 'endl' was not declared in this scope
   Serial << endl << endl;
             ^
mqtt_relay:24: error: 'Homie_setFirmware' was not declared in this scope
   Homie_setFirmware("awesome-relay", "1.0.0");
                                             ^
mqtt_relay:26: error: 'class HomieNode' has no member named 'advertise'
   lightNode.advertise("on").settable(lightOnHandler);
             ^
exit status 1
'HomieRange' does not name a type

```

Any ideas how I get things to compile ?

Lars

graugans, 2016-10-31:
Looks like  you try to build an example which is for the current master Branch.

graugans, 2016-10-31:
Please check the [issue #148](https://github.com/marvinroger/homie-esp8266/issues/148). This solution solved my issue

LarsAC, 2016-10-31:
Yep, solved - got lost in the various versions. Thanks a lot.

closed 2016-10-31 by LarsAC