homieiot/homie-esp8266 #205: Installation issue In IDE

znelbok, 2016-11-01:
OK - I'm really new to all this so I am reading the docs pretty closely.

From the Getting Started page it says 

> 1a. For the Arduino IDE
> 
>     Download the latest v2 release
>     Load the .zip with Sketch → Include Library → Add .ZIP Library
> 

The text Latest v2 release is linked to  this Github repository.  The latest release is shown as V1.5.0 - yet the text in the release for new feature says "BREAKING - require Arduino for ESP8266 >= 2.2.0: mDNS querying available instead of having to explicitely specify MQTT / OTA servers host / port (#28)"

Any release is V1.5 and its latest so I downloaded it (homie-esp8266-v1-setup.zip) and tried to add it to the Arduino IDE as indicated and I get an error.

> Arduino: 1.6.12 (Windows 7), Board: "Arduino/Genuino Uno"
> 
> Specified folder/zip file does not contain a valid library
> Specified folder/zip file does not contain a valid library

So what has gone wrong here?

robchenski, 2016-11-02:
For clarity.. From what I understand. V2 has NOT been released yet and therefore not available from the releases page whatever the link currently points to. 1.5 has been released. V2 is still in development (not released like 1.5) and the 2 can't be mixed. Different examples code too.

As for the loading the zip library, don't unzip it. Or open it. Just point the add library wizard to the zip file and you should be good to go! Also make sure you have the right versions of the plugins and esp8266-arduino ide installed.

You are very close, nearly there (-: hope that helps although I'm aware it's simple stuff I'm saying.

znelbok, 2016-11-02:
Thanks
I did not unzip it or anything.  Maybe the IDE is the wrong version.  I downloaded the latest version today.

I added the plugins as per the page successfully, just not the zip file I asked about here

leeiberry, 2016-11-03:
where can i get version 1.5 of http://setup.homie-esp8266.marvinroger.fr/  from ?

thanks :)

znelbok, 2016-11-03:
https://github.com/marvinroger/homie-esp8266/releases/tag/v1.5.0

leeiberry, 2016-11-03:
how do i use that this ?

znelbok, 2016-11-07:
OK, so I found my mistake.  I needed the source code and not the setup package (both are zip files)

closed 2016-11-07 by znelbok