homieiot/homie-esp8266 #207: send data every 50 milliseconds doesn't work. Limitation?

vephinx, 2016-11-05:
I would like to have precise control over an LED.

I edited the LightOnOff example to

```
bool lightOnHandler(String value) {
  if (value.toInt() >= 0 && value.toInt() <= 1023) {
    analogWrite(PIN_RELAY, value.toInt());
    Homie.setNodeProperty(lightNode, "on", value); // Update the state of the light
    Serial.println("Light brightness changed.");
  } else {
    return false;
  }
  return true;
}
```

Then I publish to `devices/some-light/light/on/set` every 50 millisecond, with message 0, and increasing in steps of 20 to 1020. The first 3 or so messages reach the ESP and publish back to `devices/some-light/light/on` however the rest don't make it. Is this a limitation with Homie?

Thanks!

euphi, 2016-11-06:
You are using the released version 1.5 of Homie, which uses "PubSubClient" as MQTT library. This library is not able to handle that many messages. 

The development version 2.0 uses the new "AsyncMQTTClient" that is much more powerful and should be able to handle the messages.

However, MQTT is not a realtime-protocol. You don't have any guarantee that your messages arrive in time, so there may be some unexpected behaviour.

vephinx, 2016-11-06:
It works with the development version 2.0 👍

closed 2016-11-06 by vephinx