homieiot/homie-esp8266 #208: OTA with BAD_CHECKSUM making device not responsible

camilorodegheri, 2016-11-06:
Hi,

I've been testing the new changes with OTA and base64 and I haven't been able to successfully get it to work yet. 

To report the current situation, when sending a ota (using @mrpace2 script) the device is issuing a 400 BAD_CHECKSUM, but rebooting and then not responding until hard reset.

I can provide the logs from serial and mqtt if needed.

Is there anyone working on this? Otherwise I can take a look at it and try to solve it.

marvinroger, 2016-11-14:
Can you test without base64? I think the base64 implementation is somehow broken.

mrpace2, 2016-11-14:
A pair of MQTT and serial logs would be great. Also, can you share your base64-encoded firmware and the matching firmware binary? Note that this is just for trying to help with the `400 BAD_CHECKSUM`. 

You must have sent an MD5 which is why Homie rebooted after the failed OTA attempt. Something went wrong during that reboot. This is probably unrelated to the triggering `400 BAD_CHECKSUM` issue. See #204.

marvinroger, 2016-11-14:
@camilorodegheri Closing this, as the base64 OTA issue will be tracked here: #204

> but rebooting and then not responding until hard reset

As @mrpace2 said, there's a known esp8266/Arduino issue. When you flash a firmware over Serial, calling ESP.restart() right after (which is done internally in Homie) hangs the ESP8266 the way you're experiencing it. However, if you power cycle the device after the flash, it will work restart without any issues.

So, as a rule of thumb: after you flash any sketch on the ESP8266, power cycle it once. There won't be any `ESP.restart()` issues after that. 😉

marvinroger, 2016-11-14:
@camilorodegheri but still, if you can provide the base64-encoded firmware and the matching firmware binary, it would be great.

mrpace2, 2016-11-15:
@marvinroger I was able to duplicate this issue. Now looking into it. Can you please reopen it?

marvinroger, 2016-11-15:
@camilorodegheri can you test again with the latest git rev?

marvinroger, 2017-02-07:
@camilorodegheri ?

marvinroger, 2017-02-11:
Closing as it seems to be resolved. Feel free to reopen if needed.

PR #219 (2016-11-15): Bugfix: base64 decoding
Fixed an issue decoding base64. The first base64 characters of consecutive payload chunks were not handled correctly. The first two base64 characters matter, not just the first one.

This should fix #208.
closed 2017-02-11 by marvinroger