homieiot/homie-esp8266 #209: Static IP after AP config?

cingolanifede, 2016-11-08:
Hi, is there any way to set up a static ip after the AP config? The router usually gives the same ip address, but if the router gives the esp8266 a new ip, can i configure homie to use the first ip that was given to it? I am using an app and after I set the config I store the ip in a database and the I use it to comunicate with the esp device.
Thanks.

rigorm, 2016-11-08:
Could you just set a DHCP static IP distribution list based on the ESP's MAC address ? that would be the easiest way
 --------- Original Message --------- Subject: [marvinroger/homie-esp8266] Static IP after AP config? (#209)

euphi, 2016-11-08:
There are many routers that don't support this. I have one of these, too and thinking about setting up an DHCP server on my home server that also runs mosquitto and openhab.

However, I don't see the need to do this for homie nodes (I do have the need for other devices, e.g. the printer), because homie works fine with variable IP address.

cingolanifede, 2016-11-08:
Thanks for your replies. I solved using the rigorm way. I get the MAC address an then I use the ARP table and get the ip so I can comunicate with the esp.
I close this issue #209.
Thanks again.

closed 2016-11-08 by cingolanifede