homieiot/homie-esp8266 #210: Changing Config After Setup

boilermanc, 2016-11-10:
Hey Marvin, i think i am just missing it in your documentation, but how do you update the config once you have it setup? i have the wifi working and it creates an ip on my network - that was cool. but now i need to add the mqtt information. 

thanks!

keesma, 2016-11-12:
You can send an MQTT $reset message with true to the device.
I can see the device goes back to AP mode and can connect it (Homie-xxxx).
However after I am stuck. Both the app and the webpage are stuck at "Waiting for the device".
Note am I am running v1.5 of Homie.

marvinroger, 2016-11-14:
@boilermanc with the v2 you can update the configuration over MQTT : https://homie-esp8266.readme.io/docs/ota-configuration-updates#section-configuration-updates 😉

marvinroger, 2016-11-20:
Feel free to reopen if needed.

closed 2016-11-20 by marvinroger