homieiot/homie-esp8266 #211: homie.config is not resolvable with multiple interfaces

karlp, 2016-11-10:
Before submitting your issue, make sure:

- [x ] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
nope, using master, it's the only one you have documentation for.

- [y ] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
see above

- [y ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

With both ethernet and wifi, trying to access homie.config fails as it can't be resolved by the preferred link.  Accessing the explicit 192.168.4.1 _works_ but the hard 302 to "homie.config" means it fails until I turn off my ethernet interface.

I _can_ screw around with network settings to only use certain routes on certain interfaces but it's a nasty out of box experience :(

marvinroger, 2016-11-14:
Except the access point IP is `192.168.1.1`. So yes, on your machine, `192.168.4.1` probably routes to `192.168.1.1` on the Wi-Fi interface, but it won't work on other's machine. There's unfortunately no solution to that.

karlp, 2016-11-14:
Ok, maybe I'd fiddled it to be on 192.168.4/24.   Sooo run it again...
```
#include <Homie.h>

void setup() {
  Serial.begin(115200);
  Homie_setFirmware("bare-minimum", "1.0.0"); // The "_" is not a typo! See Magic bytes
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

This fires up an AP, Homie-xxx as documented.  It assigns my wifi interface an ip in 192.168.1/24. Great.  I can ping 192.168.1.1, and curl shows it's actually the homie...
```
$ curl -v 192.168.1.1
* Rebuilt URL to: 192.168.1.1/
*   Trying 192.168.1.1...
* Connected to 192.168.1.1 (192.168.1.1) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.1.1
> User-Agent: curl/7.43.0
> Accept: */*
> 
< HTTP/1.1 302 Found
< Content-Type: text/plain
< Location: http://homie.config/ m     <<< yep, looks like homie...
< Content-Length: 0
< Connection: close
< Access-Control-Allow-Origin: *
< 
* Closing connection 0
```

So far so good.  But wifi, rightfully, has a lower priority than ethernet:
```
$ route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         OpenWrt.lan     0.0.0.0         UG    100    0        0 enp3s0
192.168.1.0     0.0.0.0         255.255.255.0   U     600    0        0 wlp0s19f2u3   <<< 600 is lower
192.168.3.0     0.0.0.0         255.255.255.0   U     100    0        0 enp3s0
192.168.124.0   0.0.0.0         255.255.255.0   U     0      0        0 virbr0
```

This means that the hard 302 redirect from an IP to a name causes it to go to my normal host resolver again.  And that is _not_ the captive portal on wifi.  (though, if asked, the captive portal will respond)
```
$ nslookup homie.config
Server:		192.168.3.1
Address:	192.168.3.1#53

** server can't find homie.config: NXDOMAIN
$ nslookup homie.config 192.168.1.1
Server:		192.168.1.1
Address:	192.168.1.1#53

Non-authoritative answer:
Name:	homie.config
Address: 192.168.1.1
```

As I said, it's the hard 302 that prevents this from working.  If I disable the ethernet interface, it will all work, but that's sucky in the extreme.  If it didn't do the 302, and just _accepted_ both homie.config and  the raw IP, it would work in both cases without any modification.

marvinroger, 2016-11-14:
Here's how it's done: first, we register the routes. So if you curl `192.168.1.1/device-info`, it'll work without `302`, so you can actually use the API. If there's no route (and, actually, there's no `/` handler), the [_onCaptivePortal()](https://github.com/marvinroger/homie-esp8266/blob/c612d4ae3a0e484b47d899645a9ea86f953e5bda/src/Homie/Boot/BootConfig.cpp#L182) method is called. We have done it this way because we don't have to store every captive portal probe URLs, we just catch everything.

So I am not sure what's your issue: you still have access to the API, if you only use the [API endpoints](https://homie-esp8266.readme.io/v2.0.0/docs/http-json-api#section-endpoints) there's no redirect. Excuse me if I don't understand you correctly.

marvinroger, 2016-11-19:
To mitigate the issue of multiple interfaces:

* The default AP IP is now `192.168.123.1`. You probably already have an interface on `192.168.1.xxx`, but not on `192.168.123.xxx` so it should help here
* The captive portal does not redirect to `http://homie.config` anymore, but to `http://192.168.123.1`.

Can you try with the latest git?

commit 0b8437c1 (2016-11-19): :sparkles: :bug: :fire: Change AP IP/subnet to 192.168.123.1 - fix #211
closed 2016-11-19 by marvinroger