homieiot/homie-esp8266 #212: question: async loops

boneskull, 2016-11-11:
Hi.

Thanks a ton for this lib.  You've really hit a sweet spot.

I didn't see a [Gitter](https://gitter.im) room or anywhere else to ask questions.

I'd like to animate a strip of WS2812B's ("NeoPixels").  For example, if the event "rainbow" was received, the strip will cycle through colors at a given speed.

For that to happen, I need to loop.  Here's the example code Adafruit provides:

```arduino
void rainbow(uint8_t wait) {
  uint16_t i, j;

  for(j=0; j<256; j++) {
    for(i=0; i<strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i+j) & 255));
    }
    strip.show();
    delay(wait);
  }
}
```

...but `delay()` is to be avoided when working with Things that listen on TCP ports.  In addition to the above, I would need to vary the speed, and perform different types of animations.

Can anyone recommend a good strategy or solution for this?  I've used [Task](https://github.com/Makuna/Task) in other settings, but not with Homie.  Is there anything within Homie itself--or a dependency--that could accomplish this?  Should I stay away from Task here, or would it be suitable?  I'm inexperienced with FP in C++, and wouldn't know how to handroll (but could learn). 

thanks
Chris

euphi, 2016-11-13:
Just get rid of the outer for loop (j) and make j static.

Call  rainbow from loop() and check if the time since last call is >= `wait`. If so, increase j and run the inner loop, if not, return.

If you need a more sophisticated timer, you can mix homie with some other lib. Due to lack of need, I haven't tried any of the many available libs for timers or tasks, except [automaton](https://github.com/tinkerspy/Automaton).

Automaton works fine with homie, but its purpose is to add support for statemachines. Timers are only one small aspect of automaton, so while automaton is a real great and helpful lib, it is too complicated for using it just for a timer.

marvinroger, 2016-11-14:
As @euphi said, just replace your `delay()` with a condition and some `millis()`. Here are some Arduino tutorials, so that you can understand easily:

* With `delay()`: https://www.arduino.cc/en/Tutorial/Blink
* Without `delay()`: https://www.arduino.cc/en/Tutorial/BlinkWithoutDelay

boneskull, 2016-11-14:
Thanks. Manually managing timers with `millis()` will be a headache, which is why I was interested in an abstraction.

marvinroger, 2016-11-14:
You can use [Metro](https://github.com/thomasfredericks/Metro-Arduino-Wiring) then, which allows pretty simple code:

```cpp
#include <Metro.h>

int rainbowCurrentColor = 0;
bool rainbowOngoing = false;

Metro rainbowMetro = Metro(); 

void startRainbow(uint8_t interval)
{
  rainbowCurrentColor = 0;
  rainbowOngoing = true;
  rainbowMetro.interval(interval);
  rainbowMetro.reset();
}

void setup()
{
}

void loop()
{
  if (rainbowOngoing && rainbowMetro.check() == 1) {
    if (j == 256) rainbowOngoing = false;
    for(i = 0; i < strip.numPixels(); i++) {
      strip.setPixelColor(i, Wheel((i + j) & 255));
    }
    strip.show();
  }
}
```

Note I haven't tested the code, but you get the idea!

boneskull, 2016-11-14:
Awesome, thanks!

closed 2016-11-14 by boneskull