homieiot/homie-esp8266 #214: Reset device failed via MQTT

AlexValois, 2016-11-12:
Hi,

When I send a reset request to MQTT for a particular device, I can see on the MQTT hub:
`$implementation/reset true
`
And just after I can see (for the same device):
`$implementation/reset false
`
On console of the NodeMCU, I can see:
 ```
ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```

And it stays frozen at this step.

When I restart manualy the NodeMCU it goes to the configuration mode successfully.
This is repeatable each time I send the reset request on the device.

Is the problem only related to my device ?

Thanks
(and one more time, I really love this Homie ESP8266 library. Really cool. Keep going) :)

AlexValois, 2016-11-12:
Same behavior when I update a configuration (in settings section of JSON file) via MQTT.
Here is the console output I get: 

```
✔ Configuration updated
Flagged for reboot
Device is idle
↻ Rebooting...
✖ MQTT disconnected
Triggering MQTT_DISCONNECTED event...
↕ Attempting to connect to MQTT...
✖ Wi-Fi disconnected
Triggering WIFI_DISCONNECTED event...
↕ Attempting to connect to Wi-Fi...

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```
and it stays frozen until I reboot the NodeMCU manually pressing the left button.
After reboot, I can see that setting has been properly modified in configuration.

AlexValois, 2016-11-12:
After a few upgrade of the firmware, I do not experience anymore the issue... don't know why...

closed 2016-11-12 by AlexValois

marvinroger, 2016-11-14:
When you flash a firmware over Serial, calling `ESP.restart()` right after (which is done internally in Homie) hangs the ESP8266 the way you're experiencing it. However, if you power cycle the device after the flash, it will work restart without any issues. :wink: