homieiot/homie-esp8266 #215: Device /set own property

stufisher, 2016-11-12:
It would be useful if a device could /set one of its own properties.

For example, say with lighting, you have a lamp that can respond to 
/light/on/set 1

via mqtt, but it also has a switch to toggle it. With the current implementation the switch can only directly set the /light/on property it cannot set /light/on/set. 

This means the current value is correctly reflected, but if the device is disconnected, then on reconnect it will read the last retained /light/on/set, which will now be out of sync. Therefore a flag or something on a node to send a /set message would then allow a retained value to be set on the server.

Does this make sense?

marvinroger, 2016-11-14:
It does! Will implement this, thanks for the idea.

marvinroger, 2016-11-14:
You can call `overwriteSetter()` on a `SendingPromise&` (before `.send()`). Can you test @stufisher?

closed 2016-11-14 by marvinroger

stufisher, 2016-11-14:
Awesome, thanks for your super quick work!

to note, if you call in one fn:
`relayNode.setProperty("on").overwriteSetter(true).send(relayState ? "1" : "0");`

then you need to set the overwrite back to false if the same node is called elsewhere (another fn):
`relayNode.setProperty("on").overwriteSetter(false).send(value);`

otherwise you'll end up in an infinite loop of
`*/on/set 1
*/on 1`

marvinroger, 2016-11-14:
My bad: https://github.com/marvinroger/homie-esp8266/commit/980798f26704a393a824e568438133ca2aa3dc5a the `SendingPromise&` is now automatically resetted to not overwrite the setter anymore, so you can remove the `overwriteSetter(false)`.

commit dfa03a7c (2016-11-14): :sparkles: Add overwriteSetter - fix #215