homieiot/homie-esp8266 #217: Get retained value

AllanMar, 2016-11-15:
I'm working on an IR Node but I need to know the last state. I was thinking about just saving them on the ESP (EEPROM/SPIFFS) but then I thought perhaps it would make more sense to load the retained value from MQTT instead...
I can't seem to determine any "clean" way to fit this into the current framework though. Any suggestions? Would this be something considered in the future?

marvinroger, 2016-11-15:
That's why `/set` topics must be sent as retained, so that you can receive it on boot. What's wrong with this approach?

marvinroger, 2016-11-15:
Oh, you mean the value on the property topic, not on the `/set` topics, and indeed the `/set` topics might be out of sync with the property value.

I can't think of a clean way to do this, just like you. I guess storing the value locally would be safer. For example, for my roller shutters (which I can set at a given percentage), I store the current percentage on the EEPROM.

AllanMar, 2016-11-15:
Yes, I was referring to the property topics. I'll just deal with it with EEPROM, sometimes the most obvious way is the best way.

Thanks and great project here!

closed 2016-11-15 by AllanMar

rigorm, 2016-11-15:
Hi Marvin,
 
Your project would be a good exemple to put on the example page.
 
maybe not Homie specific but still. Could be useful to show how to use the EEPROM/SPIFFS.. 
 
a striped down version
 
what do you think ?
 
J-F