homieiot/homie-esp8266 #220: Proposal: replace $ota/version with $implementation/ota/checksum

mrpace2, 2016-11-15:
I think md5 hashes should be mandatory with Homie OTA because they are a good safety net for different kinds of potential OTA issues (see #204, #208). In addition, an md5 hash could eliminate the need for constantly changing an "artificial" firmware version during development (and with it, the inconsistency with the `$ota` topic :-)). I suggest keeping `$fw/version`, but just for informational purpose.

A new flow could be like this:

- During startup, Homie reports the current firmware's md5 to `$fw/checksum` (in addition to `$fw/name` and `$fw/version`). OTA servers may or may not use this information for automatically scheduling OTA updates.
- OTA update:
  - OTA server sends the new checksum to `$implementation/ota/checksum`. Homie compares its own md5 hash with new value and responds:
    - `403` Forbidden (if OTA disabled)
    - `304` Not Modified (if new md5 equals that of the running firmware)
    - `202` Accepted (otherwise)
  - For `202`, OTA server sends new firmware blob to `$implementation/ota/firmware`. Either base64-encoded or binary. Homie updates firmware and reboots. Same as today.

I could volunteer for implementing this, and I could also look into the matching `homie-ota` PR.

jpmens, 2016-11-15:
I would welcome this, in particular for `eliminate the need for constantly changing an "artificial" firmware version during development`.

marvinroger, 2016-11-15:
I agree, but how do you store/calculate the current firmware MD5? We can store it after a first OTA, but what happens for the very first OTA after a Serial flash?

mrpace2, 2016-11-15:
`ESP::getSketchMD5()`. It's the exact same hash as what we use during download. I checked.

marvinroger, 2016-11-15:
Awesome! Then yes, let's implement this 😇

marvinroger, 2016-11-16:
Closing this as this is now implemented and the [docs are updated](https://homie-esp8266.readme.io/v2.0.0/docs/ota-configuration-updates).

closed 2016-11-16 by marvinroger