homieiot/homie-esp8266 #222: Open WLAN, sending config.json crashes ESP

Gulaschcowboy, 2016-11-15:
Hi Marvin,

trying to connect to an open WLAN using the Android app results in:

✔ Wi-Fi scan completed
Received captive portal request
Triggering Wi-Fi scan...
Received config request

Exception (28):
epc1=0x4000bf80 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff1040 end: 3fff15e0 offset: 01a0

>>>stack>>>
3fff11e0:  3ffe84db 3fff12b4 3fff1290 4020e191  
3fff11f0:  00000007 3fff12bc 00000007 3fff12bc  
3fff1200:  00000007 3fff12bc 3fff1290 4020e74a  
3fff1210:  00000007 3fff12bc 3ffed578 3ffefd50  
3fff1220:  00000007 3fff12bc 3fff1474 40219980  
3fff1230:  00000007 3fff12bc 3fff1474 402199cf  
3fff1240:  00000007 3fff12bc 3fff1474 40219a01  
3fff1250:  00000007 3fff12bc 00000537 4010020c  
3fff1260:  3fff12b4 3fff1474 3fff1294 4010068c  
3fff1270:  3fff04d0 3fff1474 3fff1474 3ffefd50  
3fff1280:  3fff12b4 3fff1474 3fff1470 4020ee72  
3fff1290:  3fff1200 3fff47b4 0000000f 00000000  
3fff12a0:  3fff12b4 3fff13b0 3ffef4c8 4020acde  
3fff12b0:  3ffe8638 3fff12b0 3fff12e4 3fff12b0  
3fff12c0:  3fff12c4 3fff12d4 3fff4c39 00000002  
3fff12d0:  3fff4c3e 00000000 3fff4c49 00000001  
3fff12e0:  3fff4c52 3fff134c 3fff4c34 00000007  
3fff12f0:  3fff12bc 3fff12b0 3fff12fc 3fff130c  
3fff1300:  3fff4c5c 00000002 3fff4c61 3fff131c  
3fff1310:  3fff4c73 00000001 3fff4c78 3fff132c  
3fff1320:  3fff4c7d 00000001 3fff4c82 3fff133c  
3fff1330:  3fff4c87 00000002 3fff4c90 00000000  
3fff1340:  3fff4c97 00000002 3fff4ca0 3fff135c  
3fff1350:  3fff4c57 00000007 3fff12f4 3fff1384  
3fff1360:  3fff4ca9 00000002 3fff4cae 3fff12b0  
3fff1370:  3fff1374 00000000 3fff4cb8 00000001  
3fff1380:  3fff4cc0 3fff1394 3fff4cb4 00000007  
3fff1390:  3fff136c 00000000 3fff4cc5 00000002  
3fff13a0:  3fff4ccf 3fff4d64 3fff47cc 3fff1400  
3fff13b0:  000000d0 3fff146c 3fff146c 40219980  
3fff13c0:  000000c0 00000001 3fff146c 402199cf  
3fff13d0:  3ffe84db 00000000 3fff146c 40219ca9  
3fff13e0:  3fff4f2a 000000ce 3fff146c 000000ce  
3fff13f0:  000000ce 3fff146c 3fff1460 40219d2c  
3fff1400:  3ffe007d 3fff14b0 3fff1460 40219b79  
3fff1410:  3fff04d0 00000537 00000537 4010020c  
3fff1420:  3fff04d0 000005ed 000005ed 4010020c  
3fff1430:  00000000 3fff14b0 3fff1460 4010068c  
3fff1440:  000000e0 3fff14b0 3fff4254 40219924  
3fff1450:  00000000 3fff14b0 3ffef4c8 3fff04d0  
3fff1460:  00000010 3fff14f0 00000000 000000f0  
3fff1470:  00000001 3fff422c 0000000f 00000000  
3fff1480:  00000000 00000000 00000000 40215664  
3fff1490:  00000010 3fff14f0 3fff4c34 40219980  
3fff14a0:  3fff262c 00000007 3fff14f0 402199cf  
3fff14b0:  3fff04d0 00000486 00000486 40215664  
3fff14c0:  00000003 00000001 3fff31bc 4021b87b  
3fff14d0:  00000000 00000000 3fff31bc 4020ce2e  
3fff14e0:  3fff31bc 3ffef508 3fff31bc 40215696  
3fff14f0:  00000000 00000000 00000000 40219b30  
3fff1500:  3fff31bc 3ffef508 3ffef4c8 40215729  
3fff1510:  3fff262c 0000000f 00000007 40212d84  
3fff1520:  00000000 000000ce 000000ce 00000001  
3fff1530:  00000003 000000ce 0000000f 3fff05c0  
3fff1540:  00000000 3ffef450 3ffef4c8 3fff05b4  
3fff1550:  00000001 3ffef4ec 3ffef4c8 40215923  
3fff1560:  3fffdad0 3ffef450 3ffefda4 40217e11  
3fff1570:  00000000 3fff4424 3ffefde8 40209ad0  
3fff1580:  3fffdad0 3ffef450 3ffefd50 4020b87c  
3fff1590:  4021b52c 3ffef364 3ffef3a0 3fff05b4  
3fff15a0:  3fffdad0 00000000 3fff05ac 4021b58e  
3fff15b0:  00000000 00000000 00000001 40207344  
3fff15c0:  3fffdad0 00000000 3fff05ac 4021a614  
3fff15d0:  feefeffe feefeffe 3fff05c0 40100718  
<<<stack<<<

marvinroger, 2016-11-15:
It only happens with an open network? I actually never tried.

Gulaschcowboy, 2016-11-15:
Yes, only with open networks.

marvinroger, 2016-11-15:
@Gulaschcowboy can you test with the latest git rev?

Gulaschcowboy, 2016-11-15:
Hi Marvin,

thank you, it works now! I guess this issue can be closed!?

Thanks,
Alex

closed 2016-11-15 by Gulaschcowboy

marvinroger, 2016-11-15:
Indeed, thanks for the feedback!

commit e8b2a8d6 (2016-11-15): :bug: Fix connection to open network - fix #222