homieiot/homie-esp8266 #223: HomieSetting setValidator() example

mhaack, 2016-11-15:
Hi Marvin, first really cool project.

Documentation mentions a `setValidator()` method for `HomieSetting` but I could not find any details or example. What should it return, boolean? How can I access the configured value of the setting?

Any hint or example is appreciated.

Thx
Markus

benzino77, 2016-11-16:
Check it here: [docs](https://homie-esp8266.readme.io/docs/custom-settings).
Yes it should return boolean.

marvinroger, 2016-11-16:
And there's an [example too](https://github.com/marvinroger/homie-esp8266/blob/develop/examples/CustomSettings/CustomSettings.ino)! Note I will remove the `unsigned long` parameter type, `long` is enough (allows values from -2,147,483,648 to 2,147,483,647), and if you want a value `> 0` then you can use a validator.

mhaack, 2016-11-16:
Thx for updating the sample.

closed 2016-11-16 by mhaack