homieiot/homie-esp8266 #224: Exception after "Receiving OTA firmware but not requested, skipping..."

ckrey, 2016-11-15:
Receiving OTA firmware but not requested, skipping...

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40249fbb depc=0x00000000

ctx: sys 
sp: 3ffffae0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc80:  00072198 3ffef5e0 40249fbb 40219db4  
3ffffc90:  40249fbb 00000000 3ffe9e51 40219179  
3ffffca0:  3ffe8f5b 00000034 40249fbb 4020f298  
3ffffcb0:  3fff3894 0000000f 00000004 40219274  
3ffffcc0:  00072198 3fff0038 3ffef5e0 4020fa80  
3ffffcd0:  00000005 00000000 00000020 40101faa  
3ffffce0:  3ffe9d05 40105287 3ffed768 3fff0038  
3ffffcf0:  40102cbd 3ffed768 00000000 3ffed448  
3ffffd00:  00007fff 08e25057 3ffee17c 40102e4c  
3ffffd10:  3ffea598 00000000 00000000 00000000  
3ffffd20:  3f000000 00000000 00000581 3fff2894  
3ffffd30:  7fffffff 3ffea598 3ffea598 00000000  
3ffffd40:  3ffeb92d 3fff346c 3ffffda0 00000000  
3ffffd50:  3ffeb92d 3fff3464 00000000 4021bd60  
3ffffd60:  00072198 40103024 3fffc200 4020f0ef  
3ffffd70:  3f000000 3fff3454 00000000 4021b956  
3ffffd80:  3fff06a8 00000017 00000017 40209d40  
3ffffd90:  00072198 000005b4 3fff395c 4010068c  
3ffffda0:  3fff20a4 0000032d 4020f0cc 4021bd38  
3ffffdb0:  3f000000 000005b4 3fff3954 4010068c  
3ffffdc0:  3fff3464 3fff288c 00000000 4021a4a8  
3ffffdd0:  3fff06a8 00000017 00000017 00000000  
3ffffde0:  3ffffe50 00000581 3fff3854 4021b7f1  
3ffffdf0:  00000581 00000000 00072198 00000000  
3ffffe00:  3ffeb8fa 000005b4 3fff027c 4020ad09  
3ffffe10:  00000581 00000000 00072198 00000000  
3ffffe20:  3ffeb8fa 000005b4 3fff027c 40209b35  
3ffffe30:  3fff20c4 3ffffed0 40208e74 4021b760  
3ffffe40:  3fff20a4 3ffffed0 40208e28 4021b7b8  
3ffffe50:  00000033 08e0b9d3 4010328a 00000100  
3ffffe60:  3fff03e1 3fff3854 3ffea5bc 00000001  
3ffffe70:  40225345 08de7d0a 00000000 3fff02d0  
3ffffe80:  00000001 3fff3594 3fff027c 40209b68  
3ffffe90:  3ffeef00 3fff20ec 3fff027c 40208784  
3ffffea0:  3fff34dc 02921c2c 4022991f 3fff20ec  
3ffffeb0:  3fff1fd4 3fff1fe0 00000006 3fff1de8  
3ffffec0:  3fff34dc 3fff1df0 3fff1dec 402087d7  
3ffffed0:  3fff34dc 3fff1df0 3fff1dec 40240498  
3ffffee0:  000005b4 7904a8c0 00000010 000005b4  
3ffffef0:  00000010 00000000 3ffeac49 4010730c  
3fffff00:  00000000 00000000 3ffeac65 3fff1fd8  
3fffff10:  3ffeb8d2 3fff1fe0 3fff3594 4023e99d  
3fffff20:  3fff1cfc 3fff2494 3fff2494 3ffee8b0  
3fffff30:  00000000 3fff3594 0000001c 3fff2494  
3fffff40:  3ffeb8c4 00000000 3fff3594 4023dd99  
3fffff50:  0104a8c0 00000612 00000000 0000001b  
3fffff60:  00000002 0000001a 4021e7cb 3ffed768  
3fffff70:  3ffeb89c 3fffdcc0 3ffe9db8 3ffe9db8  
3fffff80:  4021e73e 3ffed768 3fffdab0 3fff24ec  
3fffff90:  3fffdc80 00000000 3fff3594 4023280f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

Gulaschcowboy, 2016-11-15:
sounds like my last results of #204 thread

ckrey, 2016-11-15:
Yes, similar output after OTA attempt (with fresh homie-ota.py):
```
Triggering MQTT_PACKET_ACKNOWLEDGED event (packetId 32)...
✴ OTA available (checksum d5f35e99e9df82360a91879400820e98)
��� OTA started
Triggering OTA_STARTED event...
Firmware is base64-encoded
Receiving OTA firmware (1056/350736)...
<snip>
Receiving OTA firmware (350736/350736)...

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4024a353 depc=0x00000000

ctx: sys 
sp: 3ffffad0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffc70:  3ffef5e0 3fff070c 3ffffcac 40219aee  
3ffffc80:  4024a353 3ffef5e0 3ffffcb0 4010068c  
3ffffc90:  4024a3a2 00000299 00000007 4020f7a9  
3ffffca0:  000722c0 3ffef5e0 00000162 3fff34b4  
3ffffcb0:  0000000f 00000000 00000000 402199c0  
3ffffcc0:  000722c0 3fff070c 3ffef5e0 4020fd04  
3ffffcd0:  00000000 00000000 00000000 00000030  
3ffffce0:  3fff349c 0000000f 0000000d 00000001  
3ffffcf0:  00000001 00007fff 4021ee67 00000000  
3ffffd00:  ffffffff 00000000 3ffe9e51 00000000  
3ffffd10:  4021eeba 3ffed448 3fff20d4 00000000  
3ffffd20:  40000000 000720cd 000001f3 3fff2894  
3ffffd30:  00000177 000722c0 000000da 00000000  
3ffffd40:  3ffeb2b6 3fff37d4 3ffffda0 00000000  
3ffffd50:  3ffeb2b6 3fff37cc 00000000 4021bd60  
3ffffd60:  000722c0 00000000 00000000 4020f0ef  
3ffffd70:  40000000 3fff37bc 00000000 4021b956  
3ffffd80:  3fff2494 40103024 3fffc200 40209d40  
3ffffd90:  000722c0 00040000 7fffffff 00000002  
3ffffda0:  3fff26bc 00000003 4020f0cc 4021bd38  
3ffffdb0:  40000000 3ffeca24 00000110 00bad4ae  
3ffffdc0:  3fff37cc 3fff288c 00000000 3fff51fc  
3ffffdd0:  3fff51fc 3fff3b18 3fff2494 00000165  
3ffffde0:  3ffffe50 000001f3 3fff3844 4021b7f1  
3ffffdf0:  000001f3 000720cd 000722c0 00000000  
3ffffe00:  00000005 00000000 00000020 4020ad09  
3ffffe10:  000001f3 000720cd 000722c0 00000000  
3ffffe20:  3ffeb2b6 000005b4 3fff027c 40209b35  
3ffffe30:  3fff06a8 00000642 00000642 4010020c  
3ffffe40:  00000000 3ffe8498 00000001 4010068c  
3ffffe50:  00000000 099b1767 4010328a 4010731c  
3ffffe60:  3fff03e1 00000327 00000327 4010020c  
3ffffe70:  3fff3924 3fff1df4 3fff3924 3fff02d0  
3ffffe80:  00000001 3fff20a4 3fff027c 40209b68  
3ffffe90:  3fff3924 3fff1df4 3fff027c 40208784  
3ffffea0:  3fff3b14 00000000 3fff3b14 4023f49d  
3ffffeb0:  00000002 3fff1fe0 3fff027c 3fff1de8  
3ffffec0:  3fff3b14 3fff1df0 3fff1dec 402087d7  
3ffffed0:  3fff3b14 3fff1df0 3fff1dec 40240498  
3ffffee0:  000005b4 7904a8c0 00000018 000005b4  
3ffffef0:  00000018 00000000 3fff07a4 4010730c  
3fffff00:  40100000 00000000 00000064 3fff1fd8  
3fffff10:  3ffeb28e 3fff1fe0 3fff20a4 4023e99d  
3fffff20:  3fff1cfc 3fff2494 3fff2494 3ffee8b0  
3fffff30:  00000000 3fff20a4 0000001c 3fff2494  
3fffff40:  3ffeb280 00000000 3fff20a4 4023dd99  
3fffff50:  0104a8c0 00000612 00000000 0000001e  
3fffff60:  00000002 0000001a 4021e7cb 3ffed6c8  
3fffff70:  3ffeb258 3fffdcc0 3ffe9db8 3ffe9db8  
3fffff80:  4021e73e 3ffed6c8 00000000 3fff24fc  
3fffff90:  3fffdc80 00000000 3fff20a4 4023280f  
3fffffa0:  40000f49 3fffdab0 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,0)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v3de0c112
~ld
�Triggering NORMAL_MODE event...
```

ckrey, 2016-11-15:
Closing this as a duplicate to #204

closed 2016-11-15 by ckrey

Gulaschcowboy, 2016-11-15:
may I ask which ESP type you are using?

ckrey, 2016-11-15:
Wemos D1 mini clone / ESP12F