homieiot/homie-esp8266 #226: Set custom URL for "http://homie.config/"

keyeh, 2016-11-18:
Is there a way to have a custom captive portal URL?

Right now it appears the http://homie.config/ URL is hard coded
in `homie-esp8266\src\Homie\Boot\BootConfig.cpp` lines 184 & 191

Consider something like `Homie_setPortalUrl("http://setup.example.com"")`

marvinroger, 2016-11-19:
This commit: https://github.com/marvinroger/homie-esp8266/commit/0b8437c18c21662805ddfd1636d9c67f81a202ef does not redirect to `homie.config` anymore but to the IP, as some people had problems with DNS with multiple interfaces (see https://github.com/marvinroger/homie-esp8266/issues/211)

I can't come up with a better solution, but the DNS resolves every hostname to the IP, so you can ask your users to type `setup.example.com` and it will actually reach the ESP8266.

marvinroger, 2016-11-20:
Feel free to reopen if needed.

closed 2016-11-20 by marvinroger