homieiot/homie-esp8266 #227: suggestion: custom settings doc improvement

boneskull, 2016-11-19:
Hi,

The documentation example [reads as follows](https://homie-esp8266.readme.io/docs/custom-settings):

```cpp

HomieSetting<long> percentageSetting("percentage", "A simple percentage");  // id, description

void setup() {
  percentageSetting.setDefaultValue(50).setValidator([] (long candidate) {
    return (candidate >= 0) && (candidate <= 100);
  });

  Homie.setup();
}
```

However, I noticed CLion will complain about this syntax:

[image]

I figure others may run into this, and it may be a good idea to address it.

By casting to a `std::function` object, we can eliminate the warning:

```cpp
percentageSetting.setDefaultValue(50).setValidator(std::function<bool (long)>([] (long candidate) {
  return (candidate >= 0) && (candidate <= 100);
}));
```

* * *

Related question:

Having never used `std::function` before (knowingly), and not having touched C++ templates for over 15 years, I don't know a better way to eliminate the warning.  *Is* there any way to do this other than the above, given the definition of `setValidator()`?

marvinroger, 2016-11-20:
Well, I am not a C++ expert either, but `std::function<bool(T)>` and `bool (*)(long)` *are* compatible, otherwise it would not even compile. So I guess it's a CLion issue (again, I may be wrong).

I learned C++ while creating this project, so I am afraid I cannot help more than that.

boneskull, 2016-11-20:
Closing as this seems to be a CLion bug

closed 2016-11-20 by boneskull