homieiot/homie-esp8266 #229: some public functions return private types

boneskull, 2016-11-20:
Hi,

I've noticed that the `settable` function in `PropertyInterface` accepts an object of type `PropertyInputHandler`, which is seemingly private; it cannot be accessed by user code.  To avoid warnings, a user must be quite verbose when calling `settable()`:

```cpp
myNode.advertise("foo").settable(
  std::function<bool (const HomieRange &, const String &)>([] (const HomieRange &range, const String &value) {
    // stuff
  }));
```
  
Should `PropertyInputHandler` (and other similar classes) be public?

marvinroger, 2016-11-20:
Actually, it's not private, but inside the `HomieInternals::` namespace, so you can access it. But `PropertyInputHandler` is the same type as `std::function<bool(const HomieRange& range, const String& value)>`, so just like #227, I think it's a CLion issue. I am saying this, because if you try to compile your sketch with a verbose output, you'll see some `homie-esp8266` warnings, but it does not complain about types not being compatible. So I don't see why CLion would flag this as a warning.

boneskull, 2016-11-20:
Yeah, I'm seeing what you're saying.  It doesn't affect compilation (obviously it's a warning of some sort), but it's unclear where this is coming from in CLion.  Usually with their inspections, you can get more information, but none is available.  I bet it's a CMake thing.

boneskull, 2016-11-20:
Alright, so there are a handful of bugs around this issue in JetBrains' YouTrack.  Unfortunately, I don't know how to disable them (just started using CLion, but as I wrote above, this is especially strange behavior).  I'm going to close this and #227--and probably just use Atom w/ PlatformIO for now.  😝

closed 2016-11-20 by boneskull