homieiot/homie-esp8266 #230: Make the configuration AP open?

marvinroger, 2016-11-20:
Hi everyone, 

I am considering making the configuration AP open. I think having to type a password (that is a part of the SSID) is cumbersome and does not really secure anything anyway, as the project is public and open-source.

What do you think? Please use the reaction feature of GitHub to give your opinion, and don't hesitate to comment if you want to discuss it.

Thanks!

shogsbro, 2016-11-20:
Agree; on the assumption there isn't any way an already configured Homie node will enter config state just because it cannot connect to the WiFi network it is configured to use.

flaviostutz, 2016-11-21:
+1 on this

cobii, 2016-11-21:
+1 (Make it configurable: default open, when a flag in firmware is set PW)

marvinroger, 2016-11-21:
I did think about a flag, something like this:

```c++
Homie.setConfigurationApPassword("mypass");
```

I know, the function name is rather long, but I like explicit names.
And by default, the AP would not have any password. Everyone's happy! 😉

flaviostutz, 2016-11-21:
+1 on programmable password!

zieherf, 2016-11-21:
also +1 on the programmable password - should make everyone happy.

closed 2016-12-05 by marvinroger

marvinroger, 2016-12-05:
Done! Thanks for your feedbacks.

commit 6cd5276d (2016-12-05): :sparkles: Allow configurable config AP password - close #230