homieiot/homie-esp8266 #231: Homie does not show configuration page

averri, 2016-11-20:
I'm running the example https://github.com/marvinroger/homie-esp8266/blob/master/examples/HookToEvents/HookToEvents.ino in the ESP8266. 

The device enters in the AP mode, it's possible to get access to the WiFi, but the configuration page does not appear in the browser.

marvinroger, 2016-11-20:
The configuration page (`ui_bundle.gz`) has to be flashed to the SPIFFS. See [UI bundle](https://homie-esp8266.readme.io/docs/ui-bundle)

averri, 2016-11-21:
Hi @marvinroger, thanks for informing that. I have tried with no success. I can connect to the WiFi, but the captive portal is not working.

Following the instructions in the page https://github.com/marvinroger/homie-esp8266/tree/develop/data/homie, I understand that the items 1 and 2 are mutually exclusive, is that right? I'm not using the `config.json` file, as I want to use the captive portal. 

[image]

marvinroger, 2016-11-21:
When you say "the captive portal is not working", does it show something?

averri, 2016-11-21:
No,  it doesn't show anything. The browser doesn't open, and even if you open it manually and type any url, the captive portal does not appear.

marvinroger, 2016-11-21:
The captive portal does not appear, but *what* appears? A 404?

averri, 2016-11-21:
The browser says "There is no internet connection." Is Homie using DNS for the captive portal? I'm asking this because there is no DNS resolution.

averri, 2016-11-21:
It looks like the `ui_bundle.gz` is not being added to the SPIFFS. I have added a small code into the `setup` function to print the directories from SPIFFS:

```
void setup() {

    Serial.begin(115200);
    Serial.println();
    Serial.println();

    Serial.println("Directories:");
    String str = "";
    SPIFFS.begin();
    Dir dir = SPIFFS.openDir("/");
    while (dir.next()) {
        str += dir.fileName();
        str += " / ";
        str += dir.fileSize();
        str += "\r\n";
    }
    Serial.print(str);

    Homie.enableLogging(true);
    Homie.setFirmware("events-test", "1.0.0");
    Homie.onEvent(onHomieEvent);
    Homie.setup();
}
```

It prints "Directories:" and nothing else, which is an evidence that there is no `homie` directory in the SPIFFS. I have noticed that the `data` folder was in the wrong place, it should be in the root directory of the project. I have changed it accordingly, but still no success. I'm using Platformio to compile the project.

marvinroger, 2016-11-21:
Here's how it works:

* It intercepts every DNS request and responds with the server IP (`192.168.123.1`)
* If the route is not found (like, a captive portal probe), then it redirects to `http://192.168.123.1/`
* The `/` route does not exist, so [the Web UI is served](https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Boot/BootConfig.cpp#L204)

Have you read http://docs.platformio.org/en/stable/platforms/espressif8266.html?highlight=spiffs#uploading-files-to-file-system-spiffs ?

averri, 2016-11-21:
Yes, I'm using that process to upload to SPIFFS. Now, I can see the file there:

`/homie/ui_bundle.gz / 49414`

But sill not getting redirected to the web ui.

marvinroger, 2016-11-21:
What if you type directly `192.168.123.1` in your web browser?

averri, 2016-11-21:
It says:

This site can’t be reached
192.168.123.1 took too long to respond.

marvinroger, 2016-11-21:
Which version of Homie for ESP8266 do you use?

Gulaschcowboy, 2016-11-21:
@marvinroger Is "Have you read http://docs.platformio.org/en/stable/platforms/espressif8266.html?highlight=spiffs#uploading-files-to-file-system-spiffs ?" a platformio-only function?

If yes, did I miss that in the docs?

averri, 2016-11-21:
I'm using Homie versiom 1.5.0, installed from Platformio.

Good point Marvin, yes, that is a platformio-only function. It would be nice to add this information in the docs.

marvinroger, 2016-11-22:
@averri that's why, the captive portal is implemented as of 2.0.0, which is not yet released in pio as it's not yet ready. See the [Note for v1.x users](https://github.com/marvinroger/homie-esp8266#note-for-v1x-users) in the README, about where to find the docs.

@Gulaschcowboy no it's not a pio only function. 😉 You can do that from the Arduino IDE too (see http://esp8266.github.io/Arduino/versions/2.3.0/doc/filesystem.html#uploading-files-to-file-system)

averri, 2016-11-22:
Ahhh, that's it. ☺

Thanks for informing that. This is the excitement effect, people get excited with the project and start using it even before reading the entire documentation (myself included).

marvinroger, 2016-11-22:
No problems! 😃

closed 2016-11-22 by marvinroger