homieiot/homie-esp8266 #233: Homie - can't connect to WiFi network

averri, 2016-11-20:
I'm testing the HookToEvents example: https://github.com/marvinroger/homie-esp8266/blob/master/examples/HookToEvents/HookToEvents.ino

Trying to connect to the WiFi network using Windows 10, it says "Can't connect to this network".

marvinroger, 2016-11-21:
I am afraid I cannot do anything here. [The AP is spawned using the esp8266/Arduino API](https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Boot/BootConfig.cpp#L37-L38). Maybe your ESP8266 has some radio problems, but I personally never encountered this issue.

Maybe your Windows 10 complains because there' no Internet access on the ESP8266 AP?

averri, 2016-11-21:
I connected for the first time in Windows 10, but it does not connect anymore on Windows 10. It just happens on Windows 10, I can connect from Windows 7 to the same device.

Thanks.

averri, 2016-11-21:
I have discovered how to solve this. Go to **Manage WiFi settings** on Windows 10, and select **Forget Network**. After that, it is possible to connect to the network again.

[image]

closed 2016-11-21 by averri

marvinroger, 2016-11-22:
Thanks for the information. 👍