homieiot/homie-esp8266 #234: MQTT Traffic Load

Bee-Certain, 2016-11-21:
I have four NodeMCU boards running Homie v2.0.0 connected to a Raspberry Pi 3 running Mosquito.  Each Homie device is sending data back every 30 seconds.

CPU idle time as reported by top is around 98%:  Receive data as reported by bmon peaks at about 2k bytes/second and transmitted data peaks at about 800 bytes/second.  Life is good.

If I stop Mosquito, then the receive traffic jumps to 64 K bytes/second and the transmit traffic jumps to 115K bytes/second.

My guess is that the Homie devices are flooding the Pi's AP with attempts to reconnect??

I don't plan on stopping Mosquito during normal operation, but I don't want to bring the Pi to its knees if I need to stop the broker for some reason.  I'm using ssh over WiFi to do development and the connection becomes almost unusable if I stop Mosquito.

If my guess is correct about the Homie devices, then is there a way to throttle back on the retries?

Thanks,
Bob

marvinroger, 2016-11-21:
I am surprised of such a jump. Can you try, with `Mosquitto` stopped, to power on your Homie devices one by one and check if they have an impact on it? If so, it would be obvious the ESP8266s are the problem.

Bee-Certain, 2016-11-21:
I have four NodeMCU boards, each is implementing a different kind of monitor: Temperature/Humidity, Light Level, CO2, and Weight.  I'll try attaching the code and configuration file for the T/H Monitor at the end.  The MQTT Broker is running on a Raspberry Pi 3 with built-in WiFi plus an Edimax USB WiFi dongle.  For each operating condition I am reading Tx Bytes and Rx Bytes on wlan0 using bmon.  I am also reading idle time using top.  The values reported are a visual, average after staring at the screen for awhile -- not very scientific, but probably good enough.

Procedure: 
_No ESP devices, Mosquitto running:_
Tx: 500 bytes/second, Rx: 1K bytes/second, Idle: 95%, Mosquitto: Up
_Stopped Mosquitto:_
Tx: 0 bytes/second, Rx: 1k bytes/second, Idle: 98%, Mosquitto: Down
_Plugged in Temperature/humidity Monitor:_
Tx: 22K bytes/second, Rx: 13K bytes/second, Idle: 94%, Mosquitto: Down
_Plugged in Light Monitor:_
Tx: 62K bytes/second, Rx: 36K bytes/second, Idle: 90%, Mosquitto: Down
_Plugged in CO2 Monitor:_
Tx: 90K bytes/second, Rx: 51K bytes/second, Idle: 91%, Mosquitto: Down
_Plugged in Weight Monitor:_
Tx: 112K bytes/second, Rx: 65K bytes/second, Idle: 88%, Mosquitto: Down
_Started Mosquitto, all four devices running:_
Tx 200 bytes/second, Rx: 1k bytes/second, Idle: 98%, Mosquitto: Up

All four ESPs are variations of the same code, so if I have made an error in one, then it probably exists in all four.

Arduino Sketch:
[TH_for_NodeMCU_v03-DH22-Voltage.ino.txt](https://github.com/marvinroger/homie-esp8266/files/604512/TH_for_NodeMCU_v03-DH22-Voltage.ino.txt)

Raspberry Pi processes:
[image]

bmon with mosquitto running:
[image]

top with mosquitte running:
[image]

bmon with mosquitto stopped:
[image]

top with mosquitto stopped:
[image]

I use a python script and curl to send the configuration data to the ESPs:

`
                        data  = '{"name": "TH Monitor","device_id":"TH-monitor",'
                        data += '"wifi": {"ssid": "be-green-demo-02","password": ""},'
                        data += '"mqtt": {"host": "10.10.10.1","port": 1883,"base_topic": "devices/","auth": false},'
                        data += '"ota": {"enabled": true}}'
`
Thanks for any help,
Bob

marvinroger, 2016-11-22:
So yes, the Homie devices are obviously flooding the AP. Currently, it tries to reconnect immediately after disconnecting, so I'll add some delay, hope it helps. Otherwise, the problem comes from the lwIP TCP stack, and I won't be able to fix this alone (@ me-no-dev might help here, I am not mentionning him not to disturb it).

Bee-Certain, 2016-11-22:
I'll do some testing later this morning, but for now:

The event handler has WIFI_DISCONNECTED and MQTT_DISCONNECTED,  I think you are suggesting that I put a delay in MQTT_DISCONNECTED, is that correct?

I'm thinking I could do the same for WIFI_DISCONNECTED ???

For battery powered devices, I'm thinking that I could go into deep sleep for both of those events. 

If the above strategy works, then I'm not sure you should change anything in Homie.  I like the idea that it tries to immediately reconnect, but I have the option to throttle back on that behavior.  The classic software developer's refrain comes to mind, "that's not a bug, it's a feature".

On the other hand, having the delay "under the hood" in your code makes life easier for us mere mortals.  Best of both worlds would be to have delays in Homie that have sensible defaults, but could be overridden.

Thanks for your help,
Bob

flaviostutz, 2016-11-22:
Marvin,

For that kind of retry, a common practice is to retry in doubles of time from the last time to avoid flooding when an error won't be fixed in a long time. 
For example:
   - first try: 1s
   - second try: 2s
   - 4s, 8s, 16s... 1h, 2h... 6h, 12h...

Many implementations also limits max delay time to avoid getting month delayed retries...

Just a tip...

Flávio Stutz

MaxChinni, 2016-11-23:
@flaviostutz are you referring to some kind of [exponential backoff](https://en.wikipedia.org/wiki/Exponential_backoff)? I prefer it because of the randomness introduced in the retry interval choice.

marvinroger, 2016-11-30:
@Bee-Certain don't use `delay()` in your code, due to the asynchronous nature of the code, that would not work. It needs to be implemented directly inside the framework.

@flaviostutz @MaxChinni I would indeed prefer exponentiall backoff setup as such:

```
n = retry count, up to 6 (2^6 = 64)
fixed_delay = (2^n) * 1000 ms
randomized_delay = random between 0 and fixed_delay / 10
delay = fixed_delay - randomized_delay ms
```

Does it sound good?

flaviostutz, 2016-12-01:
After my current pull request gets merged (I will do the changes you asked), I have another pull request implementing the "back-off" feature (I am actually using it now).

Maybe we don't need the random part, because it normally has to do with RF or hard wire collisions so that competing devices won't retry always at the same time, endlessly causing collisions. Maybe that's not our case here. I created a special Timer class for this. I don't know how good random generation on ESP is though.

Let's evolve this when I submit the pull request.

Thanks.

marvinroger, 2016-12-01:
@flaviostutz I am not familiar with the very low-level RF stuff, so I don't really know... The random generation on the ESP [is pretty good](http://esp8266-re.foogod.com/wiki/Random_Number_Generator)

marvinroger, 2016-12-05:
@Bee-Certain There's now a retry strategy implemented by @flaviostutz, can you try?

Bee-Certain, 2016-12-05:
Sorry, I must be doing something wrong.  When I download the develop branch and replace library in the Arduino IDE, I get:
===============================================================================
Multiple libraries were found for "ArduinoJson.h"
 Used: /home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/libraries/ArduinoJson
 Not used: /home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/libraries/arduino_586072
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `readWeight()':
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:58: undefined reference to `HomieSetting<unsigned long>::get() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o:(.text._Z11loopHandlerv+0x38): undefined reference to `HomieSetting<unsigned long>::get() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `loopHandler()':
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:72: undefined reference to `HomieSetting<unsigned long>::setDefaultValue(unsigned long)'
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:72: undefined reference to `HomieSetting<unsigned long>::get() const'
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:62: undefined reference to `HomieSetting<unsigned long>::get() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `setup':
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:83: undefined reference to `HomieSetting<unsigned long>::setDefaultValue(unsigned long)'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `function<void (*)(), void>':
/home/bob/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional:2453: undefined reference to `HomieSetting<unsigned long>::setDefaultValue(unsigned long)'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `setup':
/home/bob/Documents/Development/panStamp/Gardening/ESP8266/arduino_sketches/WeightMonitor_for_NodeMCU_v03/WeightMonitor_for_NodeMCU_v03.ino:88: undefined reference to `HomieSetting<unsigned long>::HomieSetting(char const*, char const*)'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `std::_Function_base::_Base_manager<HomieNode::HomieNode(char const*, char const*, std::function<bool (String const&, HomieRange const&, String const&)>)::{default arg#1}::{lambda(String const&, HomieRange const&, String const&)#1}>::_M_init_functor(std::_Any_data&, {lambda(String const&, HomieRange const&, String const&)#1}&&, std::integral_constant<bool, false>)':
/home/bob/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional:1987: undefined reference to `HomieSetting<unsigned long>::HomieSetting(char const*, char const*)'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o: In function `function<HomieNode::__lambda1, void>':
/home/bob/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional:2453: undefined reference to `HomieSetting<unsigned long>::HomieSetting(char const*, char const*)'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o:(.rodata._ZTV12HomieSettingImE[vtable for HomieSetting<unsigned long>]+0x8): undefined reference to `HomieSetting<unsigned long>::isBool() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o:(.rodata._ZTV12HomieSettingImE[vtable for HomieSetting<unsigned long>]+0xc): undefined reference to `HomieSetting<unsigned long>::isLong() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o:(.rodata._ZTV12HomieSettingImE[vtable for HomieSetting<unsigned long>]+0x10): undefined reference to `HomieSetting<unsigned long>::isDouble() const'
sketch/WeightMonitor_for_NodeMCU_v03.ino.cpp.o:(.rodata._ZTV12HomieSettingImE[vtable for HomieSetting<unsigned long>]+0x14): undefined reference to `HomieSetting<unsigned long>::isConstChar() const'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
====================================================================

Replacing with the old library results in a good compilation.

Bee-Certain, 2016-12-05:
Sorry about the formatting.  I tried pasting as unformatted text, but no luck

flaviostutz, 2016-12-18:
@Bee-Certain,
Please, check if this problem is still happening with the latest "develop" branch version.

marvinroger, 2017-06-01:
MQTT timed retry was implemented, closing this.

closed 2017-06-01 by marvinroger

denis-obukhov, 2021-01-03:
> @Bee-Certain,
> Please, check if this problem is still happening with the latest "develop" branch version.

It fixed for me. Thanks