homieiot/homie-esp8266 #235: 2 or more Pin LigthOnOff example ino

lesjaw, 2016-11-22:
I still cant figure it out if i use the esp12 for 2 or more ON OFF light function.. can you give an ino example for 2 or more Pin that we use?

thank you

Before submitting your issue, make sure:

- [ ] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [ ] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [ ] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

marvinroger, 2016-11-22:
The [LedStrip example](https://github.com/marvinroger/homie-esp8266/blob/develop/examples/LedStrip/LedStrip.ino) does exactly that. 😄

lesjaw, 2016-11-22:
aah..i see it now.. damn!..thank you..will try asap

closed 2016-11-30 by marvinroger