homieiot/homie-esp8266 #236: Does homie-esp8266 allow two devices to discuss directly?

sheda, 2016-11-22:
Hi there,

First congrats for your impressive work!

I would like to know if there is a way to make two devices to discuss directly with each other directly without any third party tool/software/whatever that would forward message published on one topic to another.

An example may be be more explicit:
I would like a homie light-device to be switched on by a homie switch-device . 

Root subscription/publication on those device would be:
- On switch-device side:
1 MQTT publication on /homie/switch-device/switch/state -> true (possible with homie-esp8266)

- On light-device side:
1 MQTT subscription on /homie/switch-device/switch/state (not possible with homie-esp8266, due to API adding brand/device-id/ automatically )
1 MQTT subscription on /homie/light-device/light/on/set (possible with homie-esp8266)
1 MQTT publication on /homie/light-device/light/on (possible with homie-esp8266)

I am currently using the v1.5, but it seems that two workarounds are possible with the v2, so I'll move on it:
1- on v2, I saw that you recently added the broadcast channel: But I don't think this feature is designed for this kind of precise and device-to-device communication. In addtion it would wake up every device sleeping while message is obviouly useless for them..

2- on v2, I saw on issues tracking that you added recently access to MQTT clients directly (AsyncMqttClient& getMqttClient();) and also added several callback addition for OnMessage()/OnSuscribe() on AsyncMqttClient lib.
So it would be possible to add a callback "onSwitchedSwitch" called on topic  "/homie/switch-device/switch/state" change. (To Note that I think this doesn't really break the "Homie Convention")

Why does this device-to-device channel need to be a workaround instead of having an public method that would allow a device to subscribe to another device topic?

If you have question or if I was unclear please let me know...

Thanks a lot,
Sheda 

Before submitting your issue, make sure:

- [X] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [X] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [X] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

jpmens, 2016-11-22:
>I would like to know if there is a way to make two devices to discuss directly with each other

That is not possible: Homie relies on the PUB/SUB properties of MQTT and requires an MQTT broker.

sheda, 2016-11-22:
@jpmens Okay bad... explanation was unclear.

I didn't precise the use of the broker because it was implicite for me, I'll edit my question and explicite teh use of a broker.
Sorry...

sheda, 2016-11-23:
I'll simplify my question to the following:

Why homie-devices can't publish and/or suscribe topics from another homie-device so as to communicate "directly" throught only a MQTT broker? 

For example, if a device is a switch(*/homie/switch-device/switch/state*), and the other device is a lamp(*/homie/switch-device/light/on*), i would like the lamp to suscribe to */homie/switch-device/switch/state*.
Is it possbile?

n8twj, 2016-11-23:
Homie devices connect to the MQTT broker, which does allow you to subscribe
to the published events from other Homie devices...  however the MQTT
broker is a required component.

sheda, 2016-11-23:
perhaps I am missing something but this implementation of "Homie convention" on arduino (homie-esp8266) do __not__ allows to suscribe/publish to another homie-device due to the fact the library add automatically "brand" and our own "device-id" when suscribing/publishing from a device.

I am not sure but I think homie-devices suscribing to one anothers topics don't break the Homie convention compliance? So its more an issue of the homie-esp8266 library?

n8twj, 2016-11-24:
Hi Sheda -  Are you attempting to do something like this:
https://github.com/n8twj/Homie-OLED/blob/master/Homie-OLED.ino  [does not
work as expected]

From my quick scan of the code, it seems Homie does make a call to the
onMessage function from the Async MQTT library.
https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Boot/BootNormal.cpp#L659

My elementary guess would be that the HomieEvent type needs to be expanded
to send [unhandled?] mqtt messages to the invoking application for further
processing.

sheda, 2016-11-24:
Hello,
Yep that what I'm trying to do.

I also scanned the code yesterday, And we have a direct access to the client (AsyncMqttClient& getMqttClient(); ) would allow us to publish/subscribre to any topic and catch arriving messages with the onMessage() callback.

My point is that it would be interesting to the "homie-esp8266 library" to be able to let two homie devices subscribe to each other but in a simpler way like a public method with callback association :-)

>Hi Sheda -  Are you attempting to do something like this:
>https://github.com/n8twj/Homie-OLED/blob/master/Homie-OLED.ino  [does not
>work as expected]

What is the issue?

n8twj, 2016-11-24:
Another angle to this issue could be use Node-Red to manage the logic /
business rules.

n8twj, 2016-11-24:
I tried a subscription in my Homie-OLED code, but my onMessage handler
never seems to get called.

Otherwise, I am able to publish a message and display it on the OLED.

sheda, 2016-11-24:
Yeah, Marvin wrote some Node-Red nodes [here](https://github.com/marvinroger/node-red-contrib-homie) , but none seems to be used as fwd of message.

For your code it may be because you didn't registered your OnMessage Callback:
>void onMqttConnect(bool sessionPresent) {
>  mqttClient.onMessage(onMqttMessage); // <-- Add this line
>  uint16_t packetIdSub = mqttClient.subscribe("devices/0fc9bdef/temperature/Fahrenheit", 2);
>  Serial.print("Subscribing at QoS 2, packetId: ");
>  Serial.println(packetIdSub);
>}

marvinroger, 2016-11-30:
@sheda Hi!
I understand what you're trying to achieve. But I strongly believe devices should only be "puppets". They already need an external server to communicate (the MQTT broker), so I think they should be controlled by an external controller (like, as you said, Node-RED).

But you want to do is feasible with the `getMqttClient()` method since https://github.com/marvinroger/async-mqtt-client/commit/ec60ac050b22a356b5d5cf7e16b5f1bd739864e6 where multiple events handlers were enabled. This means you can add your own handlers without overwriting Homie's ones.

And yes, I know the Node-RED nodes are far from finished / documented, I want to get a proper Homie for ESP8266 2.0.0 release before working on these.

n8twj, 2016-11-30:
> marvinroger/async-mqtt-client@ec60ac0

I am using Node-RED using just the MQTT modules  (haven't even installed
the Homie Modules).

Example flow:

[{"id":"797f9cc.ece7464","type":"mqtt in","z":"5359cb7f.f2c3b4","name":"Outside
Temperature","topic":"devices/0fc9bdef/temperature/
Fahrenheit","qos":"0","broker":"a1f10935.fd9818","x":110,"y"
:242,"wires":[["fa62a7ed.f7fc28"]]},{"id":"290dffb.a59ef","type":"mqtt
in","z":"5359cb7f.f2c3b4","name":"Outside Humidity","topic":"devices/
0fc9bdef/humidity/percent","qos":"0","broker":"a1f10935.
fd9818","x":129,"y":290,"wires":[["6fffd914.223628"]]},
{"id":"fa62a7ed.f7fc28","type":"function","z":"5359cb7f.f2c3b4","name":"Convert
Output","func":"return {\n    payload: {\n      temperature:msg.payload\n
 }\n};","outputs":1,"noerr":0,"x":316,"y":242,"wires":[["
b5a9aec0.f773"]]},{"id":"6fffd914.223628","type":"function","z":"5359cb7f.f2c3b4","name":"Convert
Output","func":"return {\n    payload: {\n      humidity:msg.payload\n
 }\n};","outputs":1,"noerr":0,"x":313,"y":289,"wires":[["
b5a9aec0.f773"]]},{"id":"b5a9aec0.f773","type":"function","z":"5359cb7f.f2c3b4","name":"Gather
Wx Data","func":"var temperature = context.get('temperature') ||
msg.payload.temperature; \nvar humidity = context.get('humidity') ||
msg.payload.humidity; \nif (temperature) { context.set('temperature',
temperature); }\nif (humidity) { context.set('humidity', humidity); }\nif
(!temperature || !humidity) { return null; }\nreturn {\n    payload: {\n
   location:\"Outside\",\n      temperature:temperature,\n
 humidity:humidity\n    }\n};\n","outputs":1,"noerr":
0,"x":537,"y":240,"wires":[["3c46bb3f.772004"]]},{"id":"3c46bb3f.772004","type":"mqtt
out","z":"5359cb7f.f2c3b4","name":"Weather Display","topic":"weather/
display","qos":"0","retain":"true","broker":"a1f10935.
fd9818","x":868.5,"y":182,"wires":[]},{"id":"a1f10935.
fd9818","type":"mqtt-broker","z":"","broker":"mqtt.local","
port":"1883","clientid":"","usetls":false,"compatmode":
false,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","
willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]

This collects input from a BMP280+Homie sensor and publishes a new topic
(weather/display), which my OLED+Homie device subscribes to...

n8twj, 2016-11-30:
I am using Node-RED using just the MQTT modules  (haven't even installed
the Homie Modules).

Example flow:

```json
[{"id":"797f9cc.ece7464","type":"mqtt in","z":"5359cb7f.f2c3b4","name":"Outside
Temperature","topic":"devices/0fc9bdef/temperature/
Fahrenheit","qos":"0","broker":"a1f10935.fd9818","x":110,"y"
:242,"wires":[["fa62a7ed.f7fc28"]]},{"id":"290dffb.a59ef","type":"mqtt
in","z":"5359cb7f.f2c3b4","name":"Outside Humidity","topic":"devices/
0fc9bdef/humidity/percent","qos":"0","broker":"a1f10935.
fd9818","x":129,"y":290,"wires":[["6fffd914.223628"]]},
{"id":"fa62a7ed.f7fc28","type":"function","z":"5359cb7f.f2c3b4","name":"Convert
Output","func":"return {\n    payload: {\n      temperature:msg.payload\n
 }\n};","outputs":1,"noerr":0,"x":316,"y":242,"wires":[["
b5a9aec0.f773"]]},{"id":"6fffd914.223628","type":"function","z":"5359cb7f.f2c3b4","name":"Convert
Output","func":"return {\n    payload: {\n      humidity:msg.payload\n
 }\n};","outputs":1,"noerr":0,"x":313,"y":289,"wires":[["
b5a9aec0.f773"]]},{"id":"b5a9aec0.f773","type":"function","z":"5359cb7f.f2c3b4","name":"Gather
Wx Data","func":"var temperature = context.get('temperature') ||
msg.payload.temperature; \nvar humidity = context.get('humidity') ||
msg.payload.humidity; \nif (temperature) { context.set('temperature',
temperature); }\nif (humidity) { context.set('humidity', humidity); }\nif
(!temperature || !humidity) { return null; }\nreturn {\n    payload: {\n
   location:\"Outside\",\n      temperature:temperature,\n
 humidity:humidity\n    }\n};\n","outputs":1,"noerr":
0,"x":537,"y":240,"wires":[["3c46bb3f.772004"]]},{"id":"3c46bb3f.772004","type":"mqtt
out","z":"5359cb7f.f2c3b4","name":"Weather Display","topic":"weather/
display","qos":"0","retain":"true","broker":"a1f10935.
fd9818","x":868.5,"y":182,"wires":[]},{"id":"a1f10935.
fd9818","type":"mqtt-broker","z":"","broker":"mqtt.local","
port":"1883","clientid":"","usetls":false,"compatmode":
false,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","
willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]
```

This collects input from a BMP280+Homie sensor and publishes a new topic
(weather/display), which my OLED+Homie device subscribes to...

marvinroger, 2016-11-30:
Don't use the Homie Node-RED nodes anyway, it's not even up-to-date with the v2. Thanks for the sample flow!

sheda, 2016-11-30:
Hi,
> I understand what you're trying to achieve. But I strongly believe devices should only be "puppets". They already need an external server to communicate (the MQTT broker), so I think they should be controlled by an external controller (like, as you said, Node-RED).
I fully understand, I was more seeking about the philosophy of homie

> But what you want to do is feasible with the getMqttClient() method since marvinroger/async-mqtt-client@ec60ac0 where multiple events handlers were enabled. This means you can add your own handlers without overwriting Homie's ones.
Yap, as you saw my merge request, i choose this way to suit my needs ;)

And yes, I know the Node-RED nodes are far from finished / documented, I want to get a proper Homie for ESP8266 2.0.0 release before working on these.
No pb, the work load your are achieving is already tremendous

Thanks for your work.

@n8twj I'll will also have a look into that, I was planning to install NodeRED anyway. thx

closed 2019-04-02 by stritti