homieiot/homie-esp8266 #238: [Question] IteadSonoffButton example - use physical switch instead pushbutton

NemoN, 2016-11-23:
I have a physical switch connected to the esp8266 module and not a pushbutton.
It is possible to toggle the connected relay on every switch change?

I tried to change the code, but i'm unable to get it working (maybe debouncing issues?)

Example workflow:

- Turn physical Switch ON (PIN HIGH) -> Toggle physical relay (ON) -> MQTT status switch "on"
- MQTT pub switch "off" -> Toggle physical relay (now OFF)
- Turn physical Switch OFF (PIN LOW) > Toggle physical relay (now ON again) -> MQTT status switch "on"

marvinroger, 2016-11-30:
Of course it is possible! Have you had any success with a "normal" Arduino sketch, without Homie?

NemoN, 2016-12-02:
Got this working in the meantime. Thanks to the great framework 👍

closed 2016-12-02 by NemoN