homieiot/homie-esp8266 #239: MQTT communication issues when "device_id" not present in config (v2.0.0 develop)

flaviostutz, 2016-11-23:
Sorry posting this "develop" branch issue here, but I couldn't lose this bug's track. I just managed to workaround a messy problem here (two days debugging it!).

When I configure my device (POST http://192.168.123.1/config) without the "device_id" field in the JSON, it restarts, connects to Wifi, connects to MQTT, but then it can't manage to publish/subscribe and I get on timeout cycles of reconnections and trials of publishing of "$online" state with no success. 

I tried with two different MQTT servers on the cloud. One of them works flawslessly (test.mosquitto.org) and another on my VPS (a default configured mosquitto 1.4.10 - latest). I have problems with the later and with test.mosquitto I have no problems. I do connect/pub/sub flawslessly with my MQTT client to both servers, which drove me crazy!

The workaround solution was to define "device_id" on config.json, but I wanted to let Homie use the default device_id on its own. I don't know if there is something with the full mac address being on the default "device_id"... 

See attached tcpdump captures on the server.

flaviostutz, 2016-11-23:
[test.zip](https://github.com/marvinroger/homie-esp8266/files/608663/test.zip)

flaviostutz, 2016-11-23:
[tcpdump-capture.zip](https://github.com/marvinroger/homie-esp8266/files/608664/tcpdump-capture.zip)

flaviostutz, 2016-11-23:
When setting "device_id" in JSON with the same "default" value from Homie, I get the error. If I use just 6 chars from mac address it works OK. Maybe something related to MQTT clientId length?

flaviostutz, 2016-11-23:
I managed to connect to the same MQTT broker that gives error on Homie using the same "Homie-92389823983" client Id and it worked.

flaviostutz, 2016-11-24:
Passing by for my latest updates on this (spent hours trying to solve this!):
  - The problem is not dependant on server (my tests were different from server to server, thats why I had false positives). it is dependant on basic_topic + device_id being something larger than 30 chars
  - If I put a break on BootNormal, the problem starts to occur after some .subcribe(...) and publish(...) calls. The larger the topic name, less calls to mqttclient will work. I revised buffer allocations on Homie and it seems all right
  - I debugged async-mqtt-client and seems like the problem is not there
  - I debugged ESPAsyncTCP and had no luck finding what could cause this too...
  - I am giving up for now. Will use a smaller base_topic

marvinroger, 2016-11-30:
Thanks for the investigation. Will take a look as soon as I can. 😄

marvinroger, 2016-12-01:
I can't reproduce with `mosquitto 1.4.10` and a 50 bytes topic (and no `device_id` set). What's your ESPAsyncTCP version?

flaviostutz, 2016-12-03:
I used the latest from their master branch.

I couldn't reproduce this problem  directly with async-mqtt-client. I only have this problem when using Homie, even though I couldn't find the bug in Homie (maybe this is something regarding to the combination of things). I revised Homie's buffer allocations to see if there could be some mess there and it seems ok. 

Then I put logs on some lines to see what happens when it stops working (from a certain length of base topic it stops working). With the runtime breaking, in HomieNormal, I started commenting out some lines that interacts with mqtt for subscriptions. After commenting out X subscriptions it starts working again. The greater mqtt base name, more lines have to be commented out!

Another thing: my code was working and after I added one more "advertised" property to my sketch, it stopped working working. Commented it out and it returned to life...

I will create a test sketch for this later.

P.S. I had the same problem using both the latest version of mosquitto on my machine and using test.mosquitto.org.
Codes from async-mqtt espasynctcp and Homie are from the latest versions on github (manual checkout and compilation).

Sent from my iPhone

egon0, 2016-12-17:
Same problem here. If the base-topic is 'homie/devices' the bug exists, if i shorten the base-topic to 'homie' only, it works.

timpur, 2017-07-23:
Ive had the same problem when using a longish base topic ("homie/******/" *=char) and device id longish as well ("******-tank-guage"), just by changing the base topic to "homie/" everything works or by keeping the same base and shorter id things work, other wire it sits in a mqtt connect loop and never gets past sending $online. 

Is there a total mqtt topic length max? or something?

timpur, 2017-07-23:
I've had more experience with this bug now, after hours of trying to work out what triggers it. It seems that long topics + lots of nodes means that there are to many messages to send on initial connection. Homie says connected but never sends the mqtt messages then disconnects then true to reconnect and sometimes actually sends the messages. This is weird I know 

Background :
Running a basic test sctech
Located remotely from the server and connecting through the internet.

I'll post more as I work out more. But seems to me it's stringing to send messages if the que is to long ?

syssi, 2017-07-23:
Where is the location of you MQTT broken? It sounds like this issue (https://github.com/marvinroger/homie-esp8266/issues/340) applies to you, too.

marvinroger, 2017-07-23:
Can you test with the latest git? I made it a state machine, so if there's not enough TCP buffer available to send all packets, it will be retried until everything is sent.

timpur, 2017-07-23:
I have the broker setup in my main place and the device setup at the holiday house. im not going through a vpn that i know about ?

syssi, 2017-07-23:
I think you have the same mtu/fragmentation issue. A local broker will solve it.

timpur, 2017-09-05:
Looking into this issue again since it's back with another device. I solves this last time by using a really small device name.

This time I really want to work out what the issue it. ATM thinking it's a MTU issue. Anyone know how to change this on the Async TCP client?

syssi, 2017-09-05:
It's a fragmentation issue for MTUs not equal 1500. Usually the case if you use a VPN: https://github.com/marvinroger/homie-esp8266/issues/340

timpur, 2017-09-05:
weird thing is im not using a VPN that i know off, but my ISP is vividwireless, which is a 4G Wireless Modem. Does this change things? Also any way to fix ? Change the MTU size in the async tcp lib ?

Gulaschcowboy, 2017-09-05:
FYI: I opened a bug/request #3569 in https://github.com/esp8266/Arduino as this MTU issue *should* be fixed in SDK 2.1.0 which is not yet part of 2.4.0-rc1
See changelog https://github.com/espressif/ESP8266_NONOS_SDK/commit/90c641efe84066b47c4616ed367697a9f49f3ac5

### Lwip
...
3. Fix MTU negotiate bug;

timpur, 2017-09-05:
Thanks. so no fix currently :( ?

Gulaschcowboy, 2017-09-05:
Mhh, I wasn't able to modify ESPAsyncTCP für creating packets with smaller MSS. But I'm not a coder. There are some MSS values with 1460 in this lib, but modification of those values didn't seem to help as a workaround. Maybe someone with more insights can clarify.
Otherwise I suggest to up-vote   #3569 in https://github.com/esp8266/Arduino

timpur, 2017-09-06:
Found a solution that worked for me.

I changed the homie loop cycle to `10 ms`, not direct esp loop which it was before. I also remember seeing this in the docs some where but didn't think it would effect anything. Turns out looping slower can help in this situation .

From here
https://github.com/256dpi/arduino-mqtt

> On the ESP8266 it has been reported that an additional delay(10); after client.loop(); fixes many stability issues with WiFi connections.

Everything seems to be working in a stable matter :).

Gulaschcowboy, 2017-09-06:
Mhh, I'm running everything "single-shot" within setup() because it's battery driven. But maybe I can stretch the transmission there a bit, to get smaller packets in result. Ugly, maybe not so reliable workaround, but let's test.

Gulaschcowboy, 2017-09-07:
Result: I stretched the transmission with delays and most of the time the packets are getting smaller then. But only most of the time. Will do some more testing.

timpur, 2017-09-07:
Same on battery but might be a work around until the new sdk is uses by esp arduino and hopefully fixes the underlying tcp issue ....

Gulaschcowboy, 2017-09-07:
@timpur I tried to add some delay at various locations, but wasn't able to create smaller packets reliably. Sometimes still big packets are created, e.g. it contains all the initial  Homie infrastructure publishes/subscribes in one packet, sometimes not.
[image]

timpur, 2017-09-08:
Soz, dont know how else to help

stritti, 2020-01-01:
Closed because of no progress for long time.

closed 2020-01-01 by stritti