homieiot/homie-esp8266 #241: Any chance of a setBaseTopic fn?

stufisher, 2016-11-24:
I cant see this documented in the API, but it would be cool to be able to set the base topic from within the firmware.

I have a mix of devices from various points in v2.0 development, and my base topic is still the old devices, updating the firmware then requires me to send a config update. Plus homie-control is not quite ready for a dynamic basetopic yet, need to make a few changes!

marvinroger, 2016-11-30:
That's why Homie v2.0.0 is not yet released/marked as stable, because you cannot really rely on it being "frozen".

The problem is we want to keep firmwares "environments agnostic", which is the whole point of the `config.json`. Making the base topic configurable from the firmware would break this rule.

What's wrong with sending a config update?

stufisher, 2016-11-30:
No worries, i can work round :)

closed 2016-11-30 by stufisher