homieiot/homie-esp8266 #242: Strange reset behavior

zieherf, 2016-11-26:
I'm using the https://www.olimex.com/Products/IoT/ESP8266-EVB-BAT-BOX/open-source-hardware esp8266 with the current development homie firmware and the sketch that is shown at the end.
The sketch runs fine for a couple of hours or days respectively, and then suddenly starts to resetting itself every 15 seconds or so. I did observe that on two different devices. I couldn't get serial output until now that would give more information on the cause of reset (hope to get to that when it happens again). From what I found so far on the internet, it all points to a watchdog timer issue. Did anybody have a similar experience?

I had a similar sketch running with the stable 1.5 homie release and I didn't see experience any device resets. For the 2.0 sketch, I changed to bring down the relay in the loopHandler because the following sequence didn't work in relayToggleHandler any more:

```
  if (value == "1" && digitalRead(PIN_RELAY) == LOW) {
    digitalWrite(PIN_RELAY, HIGH);
    delay(250);
    digitalWrite(PIN_RELAY, LOW);
 }
```
The relay didn't go to HIGH and back to LOW (it stayed LOW), I suppose that is related to the asynchronious nature in homie 2.0?

**Homie 2.0.0 sketch:**
```
/*
 * Olimex esp8266 development board ESP8266 EVB-BAT 
 * Firmware for Homie 2.0.0 with basic push button functionality.
 * 
 */

#include <Homie.h>

const int PIN_RELAY = 5;
const int PIN_BUTTON = 0;
unsigned long relayLastOn = 0;

HomieNode relayNode("relay", "relay");

bool relayToggleHandler(HomieRange range, String value) {
  if (value == "1" && digitalRead(PIN_RELAY) == LOW) {
    digitalWrite(PIN_RELAY, HIGH);
    relayNode.setProperty("on").send("true");
    relayNode.setProperty("toggle").send("1");
    relayLastOn = millis();
    Serial << "Relay was toggled." << endl;
  } else {
    return false;
  }
  return true;
}

void loopHandler() {
  // bring the relay down to LOW after 250 ms, delay should not be called in any loop function
  if ( digitalRead(PIN_RELAY) == HIGH ) {
    unsigned long currentMillis = millis();
    if ( (unsigned long)(currentMillis - relayLastOn) >= 250) {
      digitalWrite(PIN_RELAY, LOW);
      relayNode.setProperty("on").send("false");
      relayLastOn = currentMillis;
    }
  }
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;

  // setup relay and button PIN
  pinMode(PIN_BUTTON, INPUT);
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);

  // Homie settings
  Homie_setFirmware("olimex-esp8266", "1.0.0");
  //Homie.disableLedFeedback();

  // reset device when pressing pin button for 5 sec (default)
  Homie.setResetTrigger(PIN_BUTTON, LOW, 5000);

  // relay only exposes a toggle property that
  // Use devices/dev-id/relay/toggle -> 1 to observe the relay
  // Use devices/dev-id/relay/toggle/set <- 1 to trigger the relay
  relayNode.advertise("toggle").settable(relayToggleHandler);
  relayNode.advertise("on");

  Homie.setLoopFunction(loopHandler);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

euphi, 2016-11-26:
I had some really strange resets in the past and most of the time it was a problem of power supply.

For example, I had an issue where an ESP8266 was stuck in a reset loop, when I resetted it by the reset input. Switching off and on the power supply worked fine. I finally found at that at a "real" reset the startup current is higher than during a "cold start" after power on. My power supply was not able to deliver a stable 3,3V during a reset.

So maybe you now have a similar problem that maybe linked to temperature of some of the device (thus it runs fine for a while).

Double check your supply and - because you are switching relais - your free wheel diodes.

zieherf, 2016-11-26:
My power supply is coming directly from [this 5V adapter](https://www.amazon.de/gp/product/B00OXCFIXU/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1) and I didn't see any issue for several months before switching from version 1.5 to the development version. 
I have now restarted (power cycle) both olimex boxes with Serial debug port activated. Until the devices start reseting I can't do much. I came across the issues you mentioned with the power supply, but I just don't think it's the root cause of my issue. 
When the first reset happens, the device runs then for 15 seconds before resetting again (very repeatable timing). The first resetting also doesn't coincide with the relay switching.

T94T, 2016-11-28:
Hi,

I have most likely the same problem. After a day or two my two devices loose connection every 15 seconds. If they reset, I don't know right now.

- Device1: ESP07 with firmware 2.0.0 installed. Own 5V 1A power supply. Connected to broker over internet.
- Device2: Sonoff with firmware 2.0.0 installed. Internal 5V power supply. Connected to broker over internet.

Each 15s broker says: "Client exceeded timeout, disconnecting." http://pastebin.com/HWJp780p

Thanks for any help.

AllanMar, 2016-11-28:
I'm having the same issue as well. Sometimes it happens more than others but eventually it starts losing MQTT connection every ~15seconds. The device itself it not resetting though so I don't think it's the same as the original issue here.

Nothing in the serial console other than:
??? MQTT disconnected
Triggering MQTT_DISCONNECTED event...
??? Attempting to connect to MQTT...
??? Attempting to connect to MQTT...
??? Attempting to connect to MQTT...
Sending initial information...

flaviostutz, 2016-11-28:
I have made a pull request on async-mqtt (https://github.com/marvinroger/async-mqtt-client/pull/31) client that improves disconnection/connection stability. Would anyone test that version to see if this problem could be solved?

Thanks.

T94T, 2016-11-28:
Thanks a lot. I will try the new code today. After that I have to wait, if it is more stable or not. Will come back soon...

AllanMar, 2016-11-28:
I tried async-mqtt (marvinroger/async-mqtt-client#31) last night and thought it had resolved it as it was relatively stable for an hour or so. Upon checking this morning though it started happening again over night and kept going.

flaviostutz, 2016-11-28:
When you say it is rebooting you mean there is an infinite loop with mqtt or the device is restarting?

I run with this pull request for 5h with no device resets. Are you sure all filtering and power source is enough (500mA recommended)? Are the inputs with pull ups?

Please send your test sketch.

Thanks.

AllanMar, 2016-11-28:
I'll post the sketch once i'm home. It just reads 2 digital pins though and sends MQTT on change.
The device doesn't reset but the MQTT disconnects every ~22 seconds (15second keepalive * 1.5). Mosquitto registers it as exceeding timeout. Usually once it starts doing this it keeps going, but it doesn't always start it right away.

I'm using the same Wemos D1 (powered by 5V USB) i always have for development so I don't believe the supply is an issue but i'll scope it later to be sure.

zieherf, 2016-11-28:
I moved the `relayLastOn = millis();` statement before the `digitalWrite(PIN_RELAY, HIGH);` statement in toggleRelayHandler and activated the Serial  debug port  - the device is running without problems so far. The resets I  had where of the following nature:
$online --> false, then reset with all properties published, online for 15 seconds and the device published $online --> false and it started all over again. I didn't have any problem with mqtt disconnect/connect.

benzino77, 2016-11-28:
@AllanMar what is your ```mosquitto.conf ``` parameter ```keepalive_interval```?

AllanMar, 2016-11-28:
@benzino77 It's not set, so default of 60s I would assume? Isn't that only for bridges? 
My understanding for clients it' uses the 1.5x the interval set by the client which Homie defaults to 10seconds (which i suspect is too short given the 5s ACK timeouts).

flaviostutz, 2016-11-28:
MQTT timeout time is defined by the connecting client. Homie's default timeout is 10s.

I saw that switching $online=true/false behavior during stress tests because of LW. On my tests it is not happening again after using the amentioned PR. Did you try the PR from async-mqtt with the newest Homie version (develop branch)?

T94T, 2016-12-01:
Hi,
I just wanted to say that my two ESPs are now running for more than two days without any timeout issues.
The pull request on async-mqtt (marvinroger/async-mqtt-client#31) seems to make things much more stable.
Thanks for all the great code out here!

marvinroger, 2016-12-01:
Kudos @flaviostutz ! 🎉

T94T, 2016-12-01:
Not only @flaviostutz! Same to you @marvinroger!

marvinroger, 2016-12-01:
Ahah, thanks! 😉

T94T, 2016-12-06:
Just for your information: My two devices now run for more than seven days without any issues!

zieherf, 2016-12-06:
Same here. My devices work properly for more than a week now.

closed 2016-12-06 by zieherf