homieiot/homie-esp8266 #245: cannot switch off an relay

gtom, 2016-11-27:
Hi,
I can't figure it out, silly problem: I'm trying to switch off an relay using the example in 2.0.0:

```
#include <Homie.h>
const int PIN_RELAY = 5;
HomieNode lightNode("wlanSW-01", "switch");
bool lightOnHandler(const HomieRange& range, const String& value) {
  if (value != "true" && value != "false") return false;
  bool on = (value == "true");
  digitalWrite(PIN_RELAY, on ? HIGH : LOW);
  lightNode.setProperty("on").send(value);
  Serial << "wlanSW ist " << (on ? "on" : "off") << endl;
  return true;
}

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;
  pinMode(PIN_RELAY, OUTPUT);
  digitalWrite(PIN_RELAY, LOW);
  Homie_setFirmware("WLAN-Switch", "0.9.2");
  lightNode.advertise("on").settable(lightOnHandler);
  Homie.setup();
}
void loop() {
  Homie.loop();
}
```
Starting the esp8266 I can read this through mosquitto:

```
# subscribed items (text after linenumber and pipe-symbol):
1:|homie/6001950fc24a/$homie -> 2.0.0
2:|homie/6001950fc24a/$implementation -> esp8266
3:|homie/6001950fc24a/$mac -> 60:01:94:AF:E6:CA
4:|homie/6001950fc24a/wlanSW-01/$type -> switch
5:|homie/6001950fc24a/wlanSW-01/$properties -> on:settable
6:|homie/6001950fc24a/$name -> WL-Switch-01
7:|homie/6001950fc24a/$localip -> 192.168.5.221
8:|homie/6001950fc24a/$stats/interval -> 60
9:|homie/6001950fc24a/$fw/name -> WLAN-Switch
10:|homie/6001950fc24a/$fw/version -> 0.9.2
11:|homie/6001950fc24a/$fw/checksum -> db621eed2fe23d5ed8b3f634cd388e83
12:|homie/6001950fc24a/$implementation/config -> {"name":"WL-Switch-01","wifi":{"ssid":"testnet"}, \
       "mqtt":{"host":"puppet.home.lan","port":1883},"ota":{"enabled":true,"host":"hum1.hometest.lan", \
       "port":9080,"path":"/ota","ssl":false}}
13:|homie/6001950fc24a/$implementation/version -> 2.0.0
14:|homie/6001950fc24a/$implementation/ota/enabled -> true
15:|homie/6001950fc24a/$online -> true
16:|homie/6001950fc24a/$stats/signal -> 70
17:|homie/6001950fc24a/$stats/uptime -> 31

```
Looks great, I think. 
Switch is on as stated in line 5. Now trying to switch it off:

```
mosquitto_pub -h puppet.hometest.lan -t homie/6001950fc24a/WL-Switch-01/on/set -m false
# this is the answer from mosquitto:
18:|homie/6001950fc24a/WL-Switch-01/on/set -> false
But homie obviously does not read the line obove, otherwise there should
#   be an acknowledgement similar to line 5?

# try different path:
mosquitto_pub -h puppet.hometest.lan -t homie/6001950fc24a/WL-Switch-01/set/on -m false
# getting this:
19:|homie/6001950fc24a/WL-Switch-01/set/on -> false
```
I think, my problem is using wrong path for publishing 'false'. I tried different combinations, does anyone have an idea for my correct path?
Thanks in advance.
Gtom

sheda, 2016-11-27:
Try node name with no uppercase => wl-switch-01

The homie convention state that:
> Node properties
> 
> homie / device ID / node ID / property: node ID is the ID of the node, which must be unique on a per-device basis, and which adhere to the ID format. property is the property of the node that is getting updated, which must be unique on a per-node basis, and which adhere to the ID format.

https://github.com/marvinroger/homie/tree/v2

On the full path you are correct, path follows schems for settable properties:

> brand/device-id/node/propertie/set

gtom, 2016-11-27:
Arghhh....
sheda pointed the right direction: Uppercase are no problems in this case, but I in the sketch above I defined the node with 'wlanSW-01'. In my mosquitto_pub command I tried to access it via 'WL-Switch-01'. But now when I use
```
mosquitto_pub -h puppet.hometest.lan -t homie/6001950fc24a/wlanSW-01/set/on  -m false
```
it works like a charm and I close this issue.
Thanks a lot for your help.
gTom

closed 2016-11-27 by gtom

marvinroger, 2016-11-30:
Thanks @sheda 
@gtom it happens sometimes 😉