homieiot/homie-esp8266 #246: Homie v1.5 on platformio and wrong documentation

elbowz, 2016-11-29:
Hi,

 if I follow the Getting started of documentation (https://homie-esp8266.readme.io/docs/getting-started):

* ```platformio lib install 555```

But this install the version 1.5 that don't run the bare-minimum sketch:
```
#include <Homie.h>

void setup() {
  Homie_setFirmware("bare-minimum", "1.0.0"); // The "_" is not a typo! See Magic bytes
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

benzino77, 2016-11-29:
You can install Homie v2 by putting this to your project properties:
```
lib_deps =  https://github.com/marvinroger/homie-esp8266.git
```

elbowz, 2016-11-29:
Great!

benzino77, 2016-11-29:
To tell the truth it can be mixed . Have a look here: [platformio lib_deps](http://docs.platformio.org/en/stable/projectconf.html#lib-deps)

elbowz, 2016-11-29:
I see! I have just started to use platformio, so I don't know this feature.

Anyway, if you follow the documentation for Homie v2, you'll get error

elbowz, 2016-11-30:
The "workaround":
```
lib_deps = Adafruit DHT Unified, https://github.com/marvinroger/homie-esp8266.git, ArduinoJson, AsyncMqttClient, Bounce2
```

marvinroger, 2016-11-30:
Homie v2 is not yet released, so stable version is still the v1.5. The v2 docs is made as if it was already released, so that as soon as the v2 code is ready we can release it.

elbowz, 2016-12-01:
Well, I have understand...anyway in the v2 docs is the only "mistake" I have found :+1: 

feel free to close the issue...thanks!

marvinroger, 2016-12-01:
Yeah I know it's kind of messy.. Sorry about that. ;)

closed 2016-12-01 by marvinroger