homieiot/homie-esp8266 #247: Upload sketch, Configuration and reset pin...lose the configuration

elbowz, 2016-11-29:
I'm starting to learn this awesome framework, but I some problems...

I would like edit, upload and try/test my sketch on esp8366 many times.

But each time I upload a new sketch I lose the configuration and I have to restart the procedure (post the JSON)...I would like don't use OTA in this phase (understanding/debugging)

My Upload sketch procedure:
* set IO0 (pin 18) to low (flash mode) 
* touch reset (pin 1) with ground...I have also try to turn off power
* upload the sketch
* set IO0 (pin 18) to high (run mode)

AllanMar, 2016-11-29:
IF you hold flash for 5seconds while it is running you will reset Homie.
See: https://homie-esp8266.readme.io/docs/resetting

You should be ok if you do reset low->hold flash->release reset. The point is to have flash held when the device resets to activate the bootloader, once that happens your fine.

elbowz, 2016-11-30:
Thanks for the answer!

Now I cannot test your solution, but hold flash don't erase the configuration (following the documentation)? I need to upload the new sketch WITHOUT lose the config...

Anyway for well understand:
* Esp8266 in normal mode
* set reset (pin 1) to LOW
* set IO0 (pin 18) to HIGH (flash mode)...about 5 seconds
* set reset (pin 1) to HIGH

Now I have ESP8266 in flash mode and I can upload the sketch, right? After? How can I go in run mode with the new sketch uploaded and with the "old" configuration?

thanks

benzino77, 2016-11-30:
There is a known problem with disappearing config on some devices. Have a look at the issue #64.
What I do is:
```
...
Homie.disableLedFeedback();
Homie.disableResetTrigger();
...
Homie.setup()
```

elbowz, 2016-11-30:
mhh....ok, I'll try.

A question: If I disable Reset ```Homie.disableResetTrigger()```, how can reset config? Only by calling ```Homie.reset();```?

benzino77, 2016-11-30:
Have a look [here](https://homie-esp8266.readme.io/docs/resetting) and [here](https://homie-esp8266.readme.io/docs/homie-implementation-specifics#section-reset).