homieiot/homie-esp8266 #249: [BUG] Timer class

elbowz, 2016-11-30:
If I have well understood the Timer lib (Timer.cpp and Timer.hpp) the Timer::reset() works as expected only when ```_tickAtBeginning == true```

I guess should be change [this line](https://github.com/marvinroger/homie-esp8266/blob/develop/src/Homie/Timer.cpp#L26) to:
```
if (tickAtBeginning) _initialTime = 0;
else this->tick();
```

flaviostutz, 2016-11-30:
Please send a sample test code showing the bug.

elbowz, 2016-11-30:
```
HomieInternals::Timer test;

void setup() {
  Serial.begin(9600);
  test.setInterval(1000, false);
}

int i = 0;
unsigned long previousMS = 0;

void loop() {
  
  if (i >= 6) {
    Serial.print(F("RESET: ")); Serial.println(i);
    test.reset();
    i=0;
  }
  
  if (test.check()) {
    Serial.print(F("CHECK: ")); Serial.print(i); Serial.print(" ("); Serial.print(millis() - previousMS); Serial.println(")");
    test.tick();
    previousMS = millis(); 
    i++;
  } 
}
```

marvinroger, 2016-11-30:
Right, will fix. Thanks!

But be careful, this is an internal component of the framework, so there's no guarantee the API won't change.

elbowz, 2016-12-01:
Thanks to you for this amazing framework! I use by 3 days but I just love it!

Why don't use a "standard" Timer lib, like you have done with Bounce lib? 
I need this feature for my little project and I think include another library could be not good.

anyway I'll use your internal library...if something change I guess will not be a big problem :+1:

closed 2016-12-01 by marvinroger

marvinroger, 2016-12-01:
Fixed, thanks!

Because that would require another dependency, which is cumbersome IMO.

commit c4192663 (2016-12-01): :bug: Fix Timer class reset - fix #249