homieiot/homie-esp8266 #250: IteadSonoffButton

Gulaschcowboy, 2016-12-02:
Hi,

after latest commits (don't ask me which one) the IteadSonoffButton example doesn't work anymore.
The ESP registers for homie/<devicename>/+/+/set, but it doesn't react to a publish on homie/<devicename>/switch/on/set

Any hints?
Thanks
Alex

elbowz, 2016-12-02:
I have found the same bug...this [commit](https://github.com/marvinroger/homie-esp8266/commit/14994b3225e51fa8a4c8430153fca76716f14729) is the problem!

This because ```device_topic``` and ```deviceId``` is not the same :)

devices/**device ID**/node-ID/property
instead the ```device_topic``` is: *node-ID/property*

```
   for (uint16_t i = 0; i < strlen(device_topic); i++) {
     if ((device_topic[i] != messageDeviceId[i]) || (device_topic[i] == '/' && messageDeviceId[i] != '\0')) {
         Interface::get().getLogger() << F("DEBUG: USCITA!!! ") << device_topic[i] << messageDeviceId[i]  << endl;
       return;
     }
   }
```

Gulaschcowboy, 2016-12-02:
@elbowz : Yes ,that's the issue. Thanks

sheda, 2016-12-03:
Hi Guys,
My bad, this issue is on me. @elbowz You are right I don't use the right topic to check the device (got confused by the name, and circumstances of my sketch made this still work for me :-/ ).
Please find fix [here](https://github.com/sheda/homie-esp8266/commit/7d6e20753c6fcdbf076d9001fd848a7d6f930c7e)
Apologies for any inconveniences.

elbowz, 2016-12-03:
Thanks for your rapidly solution! :)

marvinroger, 2016-12-05:
Merged, thanks guys!

closed 2016-12-05 by marvinroger