homieiot/homie-esp8266 #255: Node not registered

mikecay, 2016-12-06:
Hi, 
I installed the new dev repo today.  I am using the letstrip example for my node and when I send homie/plug/light/relay_1/set I get Node light not registered.  I saw people report this error back in August but the fixes do not seem to apply.  My sketch is as follows

`#include <Homie.h>

#define FW_NAME "ac_sw_os"
#define FW_VERSION "1.0.0"

/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */

const unsigned char NUMBER_OF_PLUGS = 2;
const unsigned char AC_PINS[NUMBER_OF_PLUGS] = { 12, 14 };

HomieNode plugNode("plug", "switch");

bool plugHandler(const HomieRange& range, const String& value) {
  Serial << "Range " << range.index << endl;
  if (!range.isRange) return false;  // if it's not a range

  if (range.index < 1 || range.index > NUMBER_OF_PLUGS) return false;  // if it's not a valid range

  if (value != "on" && value != "off") return false;  // if the value is not valid
  bool on = (value == "on");
  digitalWrite(AC_PINS[range.index - 1], on ? HIGH : LOW);
  plugNode.setProperty("relay").setRange(range).send(value);  // Update the state of the led
  Homie.getLogger() << "Plug " << range.index << " is " << value << endl;

  return true;
}

void setup() {
 for (int i = 0; i < NUMBER_OF_PLUGS; i++) { 
   pinMode(AC_PINS[i], OUTPUT);
   digitalWrite(AC_PINS[i], LOW);
 }

  Serial.begin(115200);
  Serial << endl << endl;
  
  Homie_setBrand("OutsideRelay"); // before Homie.setup();
  Homie_setFirmware(FW_NAME, FW_VERSION);
  plugNode.advertiseRange("relay", 1, NUMBER_OF_PLUGS).settable(plugHandler);
  
  Homie.setup();
}

void loop() {
  Homie.loop();
}`

Any help would be appreciated.

Mike

mikecay, 2016-12-06:
I noticed when my node starts my mosquitto server never receives the $nodes property, it receives all the others though.

Mike

egon0, 2016-12-17:
I could also observe this behaviour. I suppose there is a connection to #257 & #260.

marvinroger, 2017-06-01:
The `light` node is indeed not registered:

`HomieNode plugNode("plug", "switch");`

The `plug` node however, is.
So you'll have to publish to `homie/<your device id>/plug/relay_1/set`

closed 2017-06-01 by marvinroger

johnmay1, 2017-10-28:
@marvinroger, This is bug showed up occasionally and may be due to file compile sequence. more info at https://stackoverflow.com/questions/6160083/problem-with-initialize-static-stdvectorclass-in-same-class-constructor