homieiot/homie-esp8266 #256: Default Configuration / config.json Upload

KoffeinKaio, 2016-12-06:
Hey, 

I test a lot with my esp/homie module and have to reapply the config every time a new firmware gets flashed.
I tried to upload a config.json with arduino-ide espfs uploader but its always " /homie/config.json doesn't exist" (also the bin file takes ages to upload).

It there an option to specify the wifi/mqtt settings directly int he code?

robchenski, 2016-12-13:
I think that happened to me when I tried to upload config.json that way. I removed the \data\ folder and is contents from the sketch directory and reflashed without it which seemed to fix that (can't be 100% sure). But no more lost config message.

Obviously you'll need to setup config.json another way e.g. the app or webpage method.
But maybe only once (-:
I'm never certain if it's me doing something wrong or a problem with the code, just my experience.

marvinroger, 2017-01-23:
I'll test whenever I have time to.

Sorry for the long response time, exams, you know...

marvinroger, 2017-02-06:
@KoffeinKaio can you test with the web setup? http://setup.homie-esp8266.marvinroger.fr/

marvinroger, 2017-06-01:
Please try changing the SPIFFS size from your Arduino IDE, it might be too big for your hardware. I had this issue once, and changing the SPIFFS size did solve this.

closed 2017-06-01 by marvinroger