homieiot/homie-esp8266 #257: Scheduled loop handler for nodes

guibom, 2016-12-08:
_First off, Homie is awesome! I really like how easy it is to make something without having to write boilerplate MQTT code, wifi setup, etc, etc. Thanks for the hard work!_

It's a bit awkward to deal with many sensors that need to be update at different time intervals. The main Homie loop needs to keep track of elapsed time, and it just seems to go against individual handlers that settable nodes use.

I think it would be more elegant if it was possible to set individual loop handlers for nodes, with a custom interval. So basically like a scheduler, like...

```
temperatureNode.setLoopFunction(TempLoopHandler, 5000);
temperatureNode.advertise("degrees");
```

Some extra helpful functions would be a way to start/stop the schedule events and change the time interval.

NemoN, 2016-12-08:
👍

furyfire, 2016-12-08:
You can relatively easy accomplish this using the TimeAlarms.h arduino lib.
https://github.com/PaulStoffregen/TimeAlarms
I do the following

`
#include <TimeAlarms.h>
const int WEATHER_INTERVAL = 180;
void loopHandler() {
    Alarm.delay(0);
}

void setup() {
....
Homie.setLoopFunction(loopHandler);
Alarm.timerRepeat(WEATHER_INTERVAL, WeatherUpdate);
}
`
With NTP it is also possible to set the time and trigger an event at a fixed time.

flaviostutz, 2016-12-09:
I've been doing this by using Timer class from Homie core. Looking at my project this really could be embedded to the loop call, because in practice all my loop functions have to perform some kind of timing control...

+1

marvinroger, 2017-01-23:
Definitely, good idea! This will be a post 2.0 feature, as we need to focus on stability (the dev branch seems pretty buggy right now.

Sorry for the long response time, studies are my priority!

euphi, 2017-01-29:
This would be a very useful feature, so it would be really nice to have it implemented. I agree that this is too much for 2.0 and focus now should be on stabling the 2.0, so it can be officially released.

However, I don't think that 2.0 is that buggy - I think it is more that Homie is more and more known so it also attracts some novice C++ developers. Most of the bugs that make Homie seem unstable are actually caused by problems outside Homie.
The step (for the developer) from some simple playing around with an arduino to a real project with Homie and maybe some other libraries is huge.
Maybe some kind of Community Forum would be good to help Homie user and make the issue tracker focus more on real bugs again.

marvinroger, 2017-02-01:
@euphi you're right. I just created a [Gitter room](https://gitter.im/homie-iot/ESP8266), a badge on the README and a link to the room in the issue template.

tripflex, 2017-12-23:
👍  this would be great to have

timpur, 2017-12-23:
Missed this 
Since I'm going to be adding a new event bus and trimmer mechanism it would be easy to add to v2.2

timpur, 2018-01-03:
See https://github.com/marvinroger/homie/issues/53