homieiot/homie-esp8266 #258: d1 mini reset

boneskull, 2016-12-11:
Hi,

As per the docs, you can reset a device by holding the "flash" button for 5s; there is no such button on the D1 Mini.  Alternatively, it mentions something about pin 1 "low"--is this "D1" or whatever pin 1 in Arduino maps to?  Simply wiring D1 to GND doesn't seem to do the trick (but I'm also unsure *when* to do this).

I had a situation where my MQTT broker's IP changed, so I was unable to update the configuration over MQTT.  Is it possible to run the UI (via `ui_bundle.gz`) while in normal mode, just in case?

bertmelis, 2016-12-20:
pin 1 is indeed the reset pin. the board also has "RST" printed on this pin. This is also the pin which is connected to the reset button. D1 on the other hand is something else.  D1 is GPIO 5 or pin 14.

peter-clarke, 2016-12-31:
On the Wemo D1 Mini to reset; connect D3 to GND for 5 seconds.

boneskull, 2016-12-31:
@peter-clarke thanks

closed 2016-12-31 by boneskull

jimsy3, 2017-09-29:
i probably landed here via google, but what is the Reset pin number of the regular(big) D1 board in IDE software?
because some screens also have a RST pin and i need to connect the pins together, although i dont know how to map the pin in the IDE software later?

orithena, 2017-09-29:
@jimsy3: Which IDE software? Do you mean Arduino IDE?
Anyway, why would you want to use RST in your code? This pin is out of software control. To reset your board via software, put a 1K-10K between D0 and RST and control D0 for reset. If you want to wake it up from deep sleep, do the same, because D0 is the timer interrupt pin of the RTC that's still running during deep sleep. The reset button also connects to RST. So if you want to check the reason for wakeup: You're out of luck due to a hardware bug in the ESP8266 (which is why you need the external "deep sleep wake up resistor" between D0 and RST in the first place). Only way to check the wake up reason is to come up with some external parts that tell you the reason on D1-D2,D5-D7 (D3,D4,D8 are not available for that because they affect the boot mode).

bertmelis, 2017-09-29:
As @orithena already pointed out...

I believe there is a misunderstanding. The RST pins on the developement boards like the Wemos D1 (mini) is a hardware reset. The reset pin the Homie-ESP8266 firmware is referring to is a software reset pin that erases the config file and performs a soft reset. Because it is firmware controlled, you can use any suitable pin of disable it completely.
The pin labeled RST has nothing to do with the firmware. By pressing the button, the ESP restarts like you would toggle the power.

So in the Homie firmware, the software reset button is the pin with label D3 which is connected to the FLASH button and GPIO0.
On a Wemos D1 mini GPIO0 is also labeled D3 but there is no FLASH button. On the D1, the pin label depends on the board version. You can find the schematic on the Wemos website.

davidbuzz, 2017-10-28:
I'm also followin up here to point out that when programming the WeMos D1 mMini via Arduino, you should select Tools Menu -> Reset Method -> nodemcu and then the esp8266 will be automatically reset by the IDE into programming mode for you, with no buttons needing to be pressed, and no pins needing to otherwise be pulled high or low.   this is the easiest and best way to program it.

TectonicTMI, 2019-06-01:
> As @orithena already pointed out...
>
> I believe there is a misunderstanding. The RST pins on the developement boards like the Wemos D1 (mini) is a hardware reset. The reset pin the Homie-ESP8266 [...]

I appreciate you taking the time to explain this, as I also had this misunderstanding and even though I found "the right pin number" via another Github post, it didn't explain "why". Because of your (year and a half old) post, I now understand, so a belated thank you!

Hansta98, 2019-09-01:
Please again for the idiots under us (like me):
How can I fix the problem? Now since about two weeks comes the report while uploading a sketch:
Leaving...
Hard resetting via RTS pin...
But nothing happens. I just waited for hours...
I reinstalled the Arduino IDE a few times, updated any library and searched solutions in web

stritti, 2019-09-03:
> Please again for the idiots under us (like me):
> How can I fix the problem? Now since about two weeks comes the report while uploading a sketch:
> Leaving...
> Hard resetting via RTS pin...
> But nothing happens. I just waited for hours...
> I reinstalled the Arduino IDE a few times, updated any library and searched solutions in web
> At the moment I use the IDE 1.8.9

Have you tried any other sketch? I think this is an issue of Arduino IDE.

Hansta98, 2019-09-04:
Sure, I tried also to upload simple sketchs like Blink from the examples from Arduino IDE, but I come usually in the same loop 🤦‍♂️
Hard resetting....

stritti, 2019-09-04:
Maybe his page could help: https://www.arduino.cc/en/Guide/Troubleshooting

expouic, 2024-03-31:
You can connect D3 to GND and press the reset button without remove the connection. Works for me :)