homieiot/homie-esp8266 #259: question: Can I change the wifi channel within homie's config?

gregsvo, 2016-12-14:
I've got a project I'm working on at my office, but there are many, many wireless devices in the building. The problem (to my knowledge) is that the ESP8266 defaults to the heavily-used channel 1, and is causing connectivity issues. I would like to change the wifi channel to something else. Is there a way in Homie to do so?

Thank you! I'm very new at this, so I appreciate your patience.

furyfire, 2016-12-20:
You can not configure it from within Homie
But it is an easy fix to implement yourself in Homie

\src\Homie\Boot\BootConfig.cpp Line 38
```
if (Interface::get().configurationAp.secured) {
    WiFi.softAP(apName, Interface::get().configurationAp.password);
  } else {
    WiFi.softAP(apName);
  }
```

Simply change to 

```
 if (Interface::get().configurationAp.secured) {
    WiFi.softAP(apName, Interface::get().configurationAp.password, MY_CUSTOM_CHANNEL);
  } else {
    WiFi.softAP(apName, "", MY_CUSTOM_CHANNEL);
  }
```

marvinroger, 2017-01-23:
We could make it a constant in src/Homie/Constants.hpp, it would make it easier to change for anybody who need this.

Sorry for the long response time, studies are my priority!

marvinroger, 2017-06-02:
Implemented, look at this: 

```json
{
  "name": "The kitchen light",
  "device_id": "kitchen-light",
  "wifi": {
    "ssid": "Network_1",
    "password": "I'm a Wi-Fi password!",
    "bssid": "DE:AD:BE:EF:BA:BE",
    "channel": 1,
    "ip": "192.168.1.5",
    "mask": "255.255.255.0",
    "gw": "192.168.1.1",
    "dns1": "8.8.8.8",
    "dns2": "8.8.4.4"
  },
  "mqtt": {
    "host": "192.168.1.10",
    "port": 1883,
    "base_topic": "devices/",
    "auth": true,
    "username": "user",
    "password": "pass"
  },
  "ota": {
    "enabled": true
  },
  "settings": {
    "percentage": 55
  }
}
```

closed 2017-06-02 by marvinroger