homieiot/homie-esp8266 #260: setLoopFunction gets called before the connection is active

furyfire, 2016-12-14:
loopHandler gets called before Homie is successfully connected to the MQTT server

`bool first_time = true;
void loopHandler() {
    
    if (first_time) // New preset or station?
    {
        Serial.println("TOO SOON!");
        first_time = false;
    }    
}

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.println();
  Homie_setBrand("FuryFire");
  Homie_setFirmware("DeskControl", "1.0.0");
 
  Homie.setLoopFunction(loopHandler);

  Homie.setup(); 
}

void loop() {
  Homie.loop();
}`

Debug output

`↕ Attempting to connect to Wi-Fi...
TOO SOON!
✔ Wi-Fi connected
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
Sending initial information...`

Before submitting your issue, make sure:

- [X] You're using a [stable release](https://github.com/marvinroger/homie-esp8266/releases), not the Git development version which is, by definition, unstable
- [X] You've read the documentation for *your* release, which is in the `docs/` folder of the `.zip` of the release you're using, especially the **Getting started** and **Troubleshooting** pages, which contain respectively the minimum required version of the dependencies, and some answsers to the most common problems
- [X] You're using the examples bundled in *your* release, which are in the `examples/` folder of the `.zip` of the release you're using. Examples in the latest git revision might not be backward-compatible with your release

Thanks!

flaviostutz, 2016-12-14:
I think that is expected. Is documentation wrong?
You can check it on Homie.isConnected() if you wish.

furyfire, 2016-12-14:
> The loop function will be called everytime, when the device is in normal mode and the network connection is up. 

I understand that as it being run whenever the device is connected to the MQTT server

Considering other parts of the documentation says 

> If, for some reason, you want to run some code in the Arduino loop() function, it might be useful for you to know if the device is in configured (so in normal mode) and if the network connection is up.

I will sue Homie.isConnected() - Just a weird thing I'm fairly certain older versions of Homie did

marvinroger, 2017-01-23:
The loop function is meant to be called only when the connection is up, so yes, there's something wrong. I'll take a look as soon as I can.

Sorry for the long response time, studies are my priority!

elbowz, 2017-01-24:
take a look at [this commit](https://github.com/marvinroger/homie-esp8266/commit/0527756f3d19120a2b7d2ef0aa21ee8377e99b61), I had roll back and works like a charm...thanks!

furyfire, 2017-02-06:
#290 Pull request for fix

commit 7086bfdf (2017-02-06): :bug: Fix loopFunction being called too early before MQTT connection is established - closes #260 (#290)
commit 9968042d (2019-03-10): Proposal to optionally run HomieNode::loop() also in disconnected state (#490)

* First proof of concept for $broadcast feature

- $broadcast is handled and globalInputHandler is called
- still many TODOs, e.g. support for other callbacks or splitted MQTT
message (make use of payload buffer)

* $broadcast - use payload buffer

+ minor improvements

* Added $broadcast handler

+ added comments to BootNormal.cpp to explain handling of incoming MQTT
message

* Make HomieNode CONST again! :-)

* fixed wrong merge

* Added destructor to HomieNode, that will abort()

* Local develop (#3)

* :memo: A [... cut]
closed 2017-02-06 by marvinroger