homieiot/homie-esp8266 #263: More than 3 nodes not possible in 2.0.0 develop?

egon0, 2016-12-18:
At the moment i'm stuck at trying to have more than 3 nodes in the current dev-branch.
I have a BME280 for temperature, humidity and pressure and a tsl2561 light sensor for light-intensity.
If i have 3 Nodes like:

```
HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");
HomieNode pressureNode("pressure", "pressure");
```

everthing works fine. as soon as i add another node like:
```
HomieNode temperatureNode("temperature", "temperature");
HomieNode humidityNode("humidity", "humidity");
HomieNode pressureNode("pressure", "pressure");
HomieNode luminanceNode("luminance", "luminance");
```
it connects to mqtt but loses the connection some seconds later, my mosquitto log says "connection timed out".

anybody has any idea?

scottfuhrman, 2016-12-24:
Thank you for posting this. I had the exact same issue. I removed the third node and all works fine again.

Before removing the third node, I would get immediate disconnects from Mosquitto. After removing, no issues and connection is very stable.

Mosquitto logs showed:

1482538571: Client Homie-18fe34a18ccd has exceeded timeout, disconnecting.
1482538571: Socket error on client Homie-18fe34a18ccd, disconnecting.
1482538572: New connection from 70.12.xx.yy on port 8883.
1482538572: New client connected from 70.12.xx.yy as Homie-18fe34a18ccd (c1, k10, u'mosquitto').
1482538586: Client Homie-18fe34a18ccd has exceeded timeout, disconnecting.
 
Over and over...

derlucas, 2016-12-28:
I also did encounter this problem. Has anyone a clue how to fix this?

marvinroger, 2017-01-23:
It does not make sense... The underlying chip is the same, I don't get why it would work on a NodeMCU but not on a Sonoff. I just received a sonoff, I'll be able to test soon.

egon0, 2017-01-25:
It doesn't work for me on an nodemcu either.

amayii0, 2017-01-25:
Haven't tried 4 nodes but works fine with 3 (DHT22 humi+temp, LED).
Will give a try once back at home.

euphi, 2017-01-29:
Most probably there is a bug in your code somewhere else that overwrites some memory belonging to a HomieNode object.

derlucas, 2017-01-29:
You can check out this commit https://github.com/derlucas/homie-sonoff/commit/2b7e409110569b13763af06818fcfa07773d6ed9 with my Sonoff-SC Firmware. Just comment in the additional nodes.

marvinroger, 2017-01-31:
I just tried with an 8 nodes firmwares, and it works fine, so there's probably an issue somewhere else as @euphi said. I will take a closer look as soon as I have some spare time.

egon0, 2017-02-02:
seems to be fixed somehow. just updated some nodes that failed in december and they seem to work fine with at least 4 nodes. will have another try in the next days with 5 or 6 nodes.

marvinroger, 2017-02-07:
Please keep us informed so we can close the issue.

egon0, 2017-03-21:
today i tried a homie-fw with 10 homie-nodes with random data (so no real sensors/actors attached). worked in my office fine for 5 hours. but then i got home and the problem is back. maybe the other issue where someone mentioned some association with the wifi-networks around is not that wrong. in my office there are 3 or 4 wifis. in my neighborhood > 25. damn it. this bug is dragging all my development on one project down :| 

```
1490117534: New client connected from 12.34.56.78 as FF-IoT-a020a6010aa5 (c1, k300, u'micha-homie').
1490117983: Client FF-IoT-a020a6010aa5 has exceeded timeout, disconnecting.
1490117983: Socket error on client FF-IoT-a020a6010aa5, disconnecting.
```

egon0, 2017-03-21:
just installed a mosquitto in my raspberry pi in the local lan, works with ten nodes again.

bertmelis, 2017-03-21:
I all recommend you to increase the keepalive value to 300secs or something. MQTT is designed   with these high values in mind. The MQTT specification mentions "several minutes" as common practice.

Ever since I increased the keepalive I haven't seen any persisting disconnects anymore.

egon0, 2017-03-22:
The keepalive is correctly changed to 300 in my scenario. it doen't matter. i can see, that the "socket error, disconnecting" message in my broker log comes after 300 seconds, without the keepalive setting it  appears earlier. the homie-device connects, i see the connect on broker-side and via serial output. but the homie device never comes to send anything to the broker and after 300 seconds (or whatever the keepalive is configured at homie-device-side) the connection times out.

bertmelis, 2017-03-22:
@derlucas You've got blocking code in your sketch:
`temp = Serial.readString();` is blocking Homie's to do it's work correctly.
Please read [this post](http://forum.arduino.cc/index.php?topic=396450) to transform it to non blocking code. You should read untill a fixed length is received or untill you received a terminator.

@egon: Since your device does nothing at all (not sending messages nor ping), maybe you also included blockign code. Can you share your sketch?

egon0, 2017-03-22:
sure i can share the test-sketch. but there can't be a general problem with the sketch, it works in my office, it works with a local mosquitto-broker inside the lan, but does not work inside two other networks.

```
#include <Homie.h>
#include <Wire.h>

#define FW_NAME "ten test"
#define FW_VERSION "0.0.34"

/* Magic sequence for Autodetectable Binary Upload */
const char *__FLAGGED_FW_NAME = "\xbf\x84\xe4\x13\x54" FW_NAME "\x93\x44\x6b\xa7\x75";
const char *__FLAGGED_FW_VERSION = "\x6a\x3f\x3e\x0e\xe1" FW_VERSION "\xb0\x30\x48\xd4\x1a";
/* End of magic sequence for Autodetectable Binary Upload */

const int ONE_INTERVAL = 60;
const int TWO_INTERVAL = 60;
const int THREE_INTERVAL = 60;
const int FOUR_INTERVAL = 60;
const int FIVE_INTERVAL = 60;
const int SIX_INTERVAL = 60;
const int SEVEN_INTERVAL = 60;
const int EIGHT_INTERVAL = 60;
const int NINE_INTERVAL = 60;
const int TEN_INTERVAL = 60;

unsigned long lastOneSent = 0;
unsigned long lastTwoSent = 0;
unsigned long lastThreeSent = 0;
unsigned long lastFourSent = 0;
unsigned long lastFiveSent = 0;
unsigned long lastSixSent = 0;
unsigned long lastSevenSent = 0;
unsigned long lastEightSent = 0;
unsigned long lastNineSent = 0;
unsigned long lastTenSent = 0;

HomieNode oneNode("one", "temperature");
HomieNode twoNode("two", "temperature");
HomieNode threeNode("three", "temperature");
HomieNode fourNode("four", "temperature");
HomieNode fiveNode("five", "temperature");
HomieNode sixNode("six", "temperature");
HomieNode sevenNode("seven", "temperature");
HomieNode eightNode("eight", "temperature");
HomieNode nineNode("nine", "temperature");
HomieNode tenNode("ten", "temperature");

void setupHandler() {
  oneNode.setProperty("unit").send("C");
  twoNode.setProperty("unit").send("C");
  threeNode.setProperty("unit").send("C");
  fourNode.setProperty("unit").send("C");
  fiveNode.setProperty("unit").send("C");
  sixNode.setProperty("unit").send("C");
  sevenNode.setProperty("unit").send("C");
  eightNode.setProperty("unit").send("C");
  nineNode.setProperty("unit").send("C");
  tenNode.setProperty("unit").send("C");
}

void loopHandler() {
  if (millis() - lastOneSent >= ONE_INTERVAL * 1000UL || lastOneSent == 0) {
    float temperature = random(-16, 31);
    Homie.getLogger() << "Current temperature one: " << temperature << " °C" << endl;
    oneNode.setProperty("degrees").send(String(temperature));
    lastOneSent = millis();
  }
  if (millis() - lastTwoSent >= TWO_INTERVAL * 1000UL || lastTwoSent == 0) {
    float temperature = random(-16, 31);
    Homie.getLogger() << "Current temperature two: " << temperature << " °C" << endl;
[... 64 lines cut ...]
  threeNode.advertise("unit");
  threeNode.advertise("degrees");
  fourNode.advertise("unit");
  fourNode.advertise("degrees");
  fiveNode.advertise("unit");
  fiveNode.advertise("degrees");
  sixNode.advertise("unit");
  sixNode.advertise("degrees");
  sevenNode.advertise("unit");
  sevenNode.advertise("degrees");
  eightNode.advertise("unit");
  eightNode.advertise("degrees");
  nineNode.advertise("unit");
  nineNode.advertise("degrees");
  tenNode.advertise("unit");
  tenNode.advertise("degrees");
  Homie.disableLedFeedback();
  Homie.getMqttClient().setKeepAlive(300);
  Homie.setup();
}

void loop() {
  Homie.loop();
}
```

egon0, 2017-03-22:
also, if there were any blocking components, i shouldn see the serial output for the ten nodes every 60 seconds i think.

marvinroger, 2017-06-01:
I identified the cause: let's keep https://github.com/marvinroger/homie-esp8266/issues/342 (same issue) for reference.

marvinroger, 2017-06-01:
Please update your https://github.com/marvinroger/async-mqtt-client (fixed there)
Some messages still might get lost, but at least it won't disconnect anymore. Please follow the #342 issue for updates

closed 2017-06-01 by marvinroger