homieiot/homie-esp8266 #264: enhancement: custom setting array

boneskull, 2016-12-18:
I'm writing some code that controls *n* relays. Each relay has the following properties:

- "NC" or "NO" (probably just a `bool`)
- Pin

It should be possible to persist custom user settings for the number of relays, and the above properties for each.

In JSON, it could look something like this:

```json
{
  "settings": {
    "relays": [
      {
        "pin": 1,
        "normally_open": true
      },
      {
        "pin": 2,
        "normally_open": false
      }
    ]
  }
}
```

Is this feasible, or a direction you wish to go in?

I can probably work around this by using a `const char*` setting w/ delimiters--or even storing JSON itself!

keyeh, 2016-12-19:
I've been storing raw escaped JSON and parsing it on the ESP with ArduinoJSON

You many need to increase `MAX_JSON_CONFIG_ARDUINOJSON_BUFFER_SIZE ` in `Limits.hpp` for longer strings.

But it's probably better to store large configurations in SPIFFs manually, I'm investigating using a settable HomieNode over MQTT.

marvinroger, 2017-01-23:
@boneskull the JSON part of the framework is one of the heaviest part in terms of RAM usage. An array means another sublevel for settings, which can already be pretty big. I'd reconsider with an ESP32, but I think the ESP8266 is not the right target for such an "high level" feature. The framework is already pretty high level, and with the upcoming TLS support I guess we'll reach the limit for the framework to remain stable.

Actually I'd use a string with delimiters... Lightweight and efficient!

closed 2017-06-02 by marvinroger